SolidEva/PrawnOS #182: Links to PrawnOS website are missing TLS

JeremyRand, 2020-06-13:
The links to the PrawnOS website (both in the GitHub repo metadata and the source code) point to `http://www.prawnos.com`; this should be changed to `https://www.prawnos.com` to enable TLS.

SolidHal, 2020-06-14:
both redirect to the github page at this point, which enforces TLS/SSL, so the links are not really wrong. If, in the future, the links point somewhere else like a full PrawnOS webpage, ssl will be enabled/enforced there as well, redirecting the http links to https. 

Unless I'm misunderstanding something?

austin987, 2020-06-15:
While it usually works, it depends on the website to keep the redirect working, and some plugins/etc. may not handle it well (I've seen that happen on other sites, but not github).

IMO it make sense to be explicit and advertise https everywhere it's supported. That's particularly important for a project that advertises security ;).

closed 2020-06-19 by SolidHal

JeremyRand, 2020-06-20:
> both redirect to the github page at this point, which enforces TLS/SSL, so the links are not really wrong. If, in the future, the links point somewhere else like a full PrawnOS webpage, ssl will be [...]

@SolidHal That's not how it works; relying on an HTTP to HTTPS redirect makes you vulnerable to sslstrip attacks.

The repo metadata still links to HTTP, can that be fixed?

SolidHal, 2020-06-20:
>The repo metadata still links to HTTP, can that be fixed?
I think I changed what you were asking? 

TIL about sslstrip, thats clever.

JeremyRand, 2020-06-20:
> I think I changed what you were asking?

Yep, metadata is fixed now, thanks!  :)

PR #184 (2020-06-19): misc: use https:// for prawnos.com links
Fixes #182