SolidEva/PrawnOS #190: Bring in c100p support and ec script

SolidHal, 2020-07-15:
https://github.com/rk-zero/PrawnOS/commit/9623dc25fea90e91574cbfdd194442d01362c940

SolidHal, 2020-07-18:
Looking for folks with a c100p to test out support

Maccraft123, 2020-07-19:
I can test on my C100PA

colemancda, 2020-10-23:
I have a C100PA and the touchscreen works great with the current release (PrawnOs Shiba v1.03)

James-h143, 2022-02-27:
i have C100P, i have "Error: No Wi-Fi device found." when running the nmcli command on the readme

realdaly, 2022-09-04:
I have c100p and I cannot connect to wifi, the network window says no network devices
my version is PrawnOS Shiba v1.3.2, Jul 28, 2022

SolidHal, 2022-09-05:
What wireless adapter are you using? Does anything show up under lsusb?

realdaly, 2022-09-10:
I guess I was using an unsupported wireless adapter