SolidEva/PrawnOS #193: Package arm specific programs

SolidHal, 2020-07-24:
hantro enabled vdpau/vaapi decoder
mojoshader for running xna games
and maybe xonotic compiled for arm, mesa-git weekly build for panfrost improvements

Maccraft123, 2020-07-31:
It should be noted that Terraria(and perhaps more XNA games) doesn't run with recent Mono version.

https://github.com/rfht/fnaify/issues/16
Rejected pull request to fix it: https://github.com/mono/corert/pull/33

Maccraft123, 2020-08-12:
Since Debian has good multiarch support, Box86 can be elevated to run traditional x86 programs like video games on arm hardware.
This requires armhf chroot on arm64 host and x86 libraries to be installed.
The Mali GPU is more than enough to run basic games, in my testing i found it to be as fast, if not faster than Intel GMA 4500M HD present on Montevina ThinkPads