SolidEva/PrawnOS #195: package and include cmt touchpad driver for x11

SolidHal, 2020-08-04:
linux touchpad experience is still less than great. chromiumos uses cmt, which offers a much nicer experience. 

the porting work has already been done: 

> 
> I've ported the chromium OS cmt driver to Linux. Here are my relevant repos:
> https://github.com/hugegreenbug/libevdevc
> https://github.com/hugegreenbug/libgestures
> https://github.com/hugegreenbug/xf86-input-cmt
> 
> libevdev should be compiled first, followed by libgestures, and finally xf86-input-cmt. I put the base and the gtest libraries in libgestures, so that is why I don't have separate libraries for them.
> 
> I'm currently packaging these for Ubuntu. Here is my ppa: https://launchpad.net/~hugegreenbug/+archive/cmt . At this moment, I'm working on the xf86-input-cmt package, but I have the other two built.
> 
> Most of the work was putting all the code together, but I also made changes for two finger scrolling and logging errors that were showing in the xorg log.

this issue from crouton documents their process: https://github.com/dnschneid/crouton/issues/244

Maccraft123, 2020-08-06:
Did you test with libinput driver? In Wayland i'm getting better experience than in Xorg

SolidHal, 2020-08-08:
I have tried libinput on xorg, its actually the default for a prawnos install. Still is less than great unfortunately
I plan on including wayland support out of the box eventually, but would like to offer a decent experience on xorg as well

Maccraft123, 2020-08-12:
I haven't found Xorg to be fast recently, with different settings for fbdev and modesetting drivers

satmandu, 2020-08-25:
FYI, I'm using ubuntu on pixel slate on xorg, with the xf86-input-cmt drivers and the evdevc and libgesture dependencies freshly compiled.

I don't seem to have two finger scrolling working, but otherwise the trackpad works better than in Wayland (since Wayland uses the plain old libinput driver.)

Xorg seems fast on this, but if any of you figure out a way to get two finger scrolling working...

espoelstra, 2020-12-14:
I'll give another vote for libinput as you can sometimes borrow the config files directly from ChromeOS that are pre-"calibrated" for specific ChromeOS devices and you can define "quirks" for devices that have known weirdness with their pressure sensitivity etc. The cmt driver is OK, but using libinput + gestures has been really nice in my experience.

SolidHal, 2020-12-15:
I had opened this before I had worked with Wayland much. I have found libinput with wayland to be a very pleasant experience vs on xorg, so I will likely stick with libinput and drop this work. 

I'll probably end up closing this issue, but will keep it around for now.