SolidEva/PrawnOS #196: gru-kevin (and likely gru-bob) need a display port firmware

SolidHal, 2020-08-04:
on boot I see 
```
fec00000.dp: Missing free firmware (non-Free firmware loading is disabled)
```
this is due to the blob dptx.bin aka `Rockchip cdn DisplayPort controller firmware`

test if video out works anyway, if it doesn't we will need an open source version of the firmware. 
commit that added the firmware to the kernel: https://lwn.net/Articles/700405/

gregordinary, 2020-10-28:
I tried video output with two adapters, neither worked in PrawnOS (XFCE, default install). 

Test using video over USB-C:
- ChromeOS: Works
- PrawnOS: No Video Output

Test using [USB-C -> USB/HDMI/Ethernet Dock](https://www.cnx-software.com/wp-content/uploads/2020/07/PinePhone-PostMarket-USB-C-Dock.jpg)
- ChromeOS: Works
- PrawnOS: No Video Output

SolidHal, 2020-10-28:
Thanks @gregordinary 
Unfortunately I am not surprised