SolidEva/PrawnOS #219: /etc/selinux/targeted/policy/policy.31 missing from default install

Maccraft123, 2020-09-19:
/etc/selinux/targeted/policy/policy.31 file is missing from default install, and it makes selinux write to tty, ignoring loglevel
This blocks https://github.com/SolidHal/PrawnOS/pull/215
A workaround was to disable it, but some people like to use it
@gdallasdye, maybe you know how to fix it?

gdallasdye, 2020-09-19:
If I remember correctly, the package policycoreutils should provide that file. Doing a quick build now, cloned from your branch, with policycoreutils added to the shared package-list under base_debs_install. For what it's worth, that package isn't provided with a default install of Debian on x86 either. 

If that doesn't do it, it may be worth installing either setools-gui or policycoreutils-gui. One of those (might) provide the same gui Red Hat uses in their distro, which I'm more familiar with, and helps with creating policy file that tells selinux that such an action is acceptable.

A quick build takes about two hours btw.

Maccraft123, 2020-09-19:
Thanks for help!
I know nothing about SELinux, so i'm leaving it up to you.
If you want to talk i'm macc24 on Freenode IRC.

gdallasdye, 2020-09-19:
This warrants a post instead of an edit. File '/etc/selinux/targeted/policy/policy.31' seems to be provided by package 'selinux-policy-targeted' that exists in either an Enterprise Linux (or a clone) or Fedora. I see it on my Oracle Linux 8 chromebook. Going to download [the noarch package](https://centos.pkgs.org/8/centos-baseos-armhfp/selinux-policy-targeted-3.14.3-20.el8.noarch.rpm.html) from here to copy and paste that file into a resource file (or files?) to be included in the initramfs. Going to edit buildInitramFS.sh and create the '/etc/selinux/targeted/policy/' directory, copy relevant file into the appropriate folder, and document the changes. Once there is something worth documenting I'll let you all know.

In short, I think all the changes can be self contained to only the initramfs. I notice vanilla Debian does not dictate selinux policy, while Enterprise Linux does. I want to respect that design as much as possible, as I have seen backups of conversations where people where using PrawnOS as an installer for their preferred os.

Maccraft123, 2020-09-19:
Is it possible to make it smaller? Putting 8mb file is not good...

rk-zero, 2020-09-20:
>  as I have seen backups of conversations where people where using PrawnOS as an installer for their preferred os.

no wonder armhf is sold out everywhere.

SolidHal, 2020-09-23:
Seeing as we do not currently use any selinux policies, and I would have to do a lot of research to determine if and how PrawnOS would implement SELinux I am going to disable selinux in the kernel as a seperate commit citing this issue and the PR for the splash. 
If you figure out something @gdallasdye we can then re-enable selinux in the kernel.

gdallasdye, 2020-09-24:
Please note I had most of this written earlier, and don't want it to go to waste. Still working on the selinux issue, or why it springs up even when U DAC was the default module, and with selinux set to permissive...

Regarding kernel config, the two major changes there are: 1) changing the initramfs filename and compression settings, and 2) setting the Default console loglevel from 7 to 1. Am I correct?

And some notes.

I've also noticed that @SolidHal's repo has had "CONFIG_SECURITY_SELINUX=y" and "CONFIG_DEFAULT_SECURITY_SELINUX=y" set for years, which I believe is consistent with Debian x86, and it's [SELinux Documentation](https://wiki.debian.org/SELinux). Debian allows for installing selinux userspace packages after install time with their offered kernel. 

Also worth noting is that on @Maccraft123's config file that "CONFIG_LSM_MMAP_MIN_ADDR=32768" is missing and 2) "CONFIG_DEFAULT_SECURITY_DAC" used to be set to "Y" (yes). Which leads to 3) SELinux would still complain about the titular missing file.

I'm also considering playing around with "mknod /dev/fb0 c 29 0" in the initramfs, and (finally) setting up a rescueshell partition.

As a contingency, would you be willing to isolate the changes that add the daemon for disabling keyboard and touchpad into a separate branch and pull request. 

And on the topic of contingencies, I've been looking at [bannerd](https://github.com/alukichev/bannerd), which appears to draw from an image file to /dev/fb0. It can read a .BMP file and draw it in the center of the screen. That would mean only needing one image file too. [And this fork](https://github.com/sker65/bannerd) has changes that let it exit when it senses a specific program running. I'm thinking if told to kill bannerd when "*dm" is running should work for both gdm and lightdm.