SolidEva/PrawnOS #222: When a new release?

duvh93, 2020-10-15:
Hello!

Are there going to be new releases in the foreseeable future? Shall I build PrawnOS from sources myself?

Maccraft123, 2020-10-15:
Yes

SolidHal, 2020-10-20:
There will be a release `Soon`. In the meantime, you can grab a daily from the github CI for any commit if you'd like. 
The main item holding back a release is packaging. PrawnOS specific modifications are currently difficult to upgrade without a full reinstall. I would like to get all of the filesystem items packaged and uploaded to the repo so that a user can upgrade portions of the system as need be. 
I would also like to get panfrost/mesa >=20.2 working, so I am currently focusing on (re)packaging panfrost/mesa for debian 10.

SolidHal, 2020-10-29:
Now!

closed 2020-10-29 by SolidHal