SolidEva/PrawnOS #223: sudo TARGET=armhf make image fails on latest version of Debian Buster

graves, 2020-10-18:
`+ chroot /home/tg/build/PrawnOS/tmp.EV9QwG6Rqy apt install -y flashmap mosys

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package flashmap
E: Unable to locate package mosys
+ cleanup
+ set +e
+ umount -l /home/tg/build/PrawnOS/tmp.EV9QwG6Rqy
+ rmdir /home/tg/build/PrawnOS/tmp.EV9QwG6Rqy
+ losetup -d /dev/loop0
+ umount -l /home/tg/build/PrawnOS/tmp.EV9QwG6Rqy
+ rmdir /home/tg/build/PrawnOS/tmp.EV9QwG6Rqy
+ losetup -d /dev/loop0
+ rm -rf /home/tg/build/PrawnOS/PrawnOS-Shiba-armhf.img-BASE
+ echo 'FILESYSTEM BUILD FAILED'
FILESYSTEM BUILD FAILED
+ exit 1
make[1]: *** [/home/tg/build/PrawnOS/filesystem/makefile:60: /home/tg/build/PrawnOS/PrawnOS-Shiba-armhf.img-BASE] Error 1
make[1]: Leaving directory '/home/tg/build/PrawnOS'
make: *** [makefile:99: image] Error 2
`

Platform: Linux debian 4.19.0-11-amd64 #1 SMP Debian 4.19.146-1 (2020-09-17) x86_64 GNU/Linux

Currently we are unable to build an image using the exact instructions from the README.

rk-zero, 2020-10-19:
The flashmap and mosys packages come from the distribution-repository deb.prawnos.com

I don't know if they are missing there, but you can simply remove the packages in:

https://github.com/SolidHal/PrawnOS/blob/11f372958a6f368d95ddcb09692faa42cd6b1b20/scripts/Shared/package_lists.sh#L183

graves, 2020-10-19:
This is the problem:

```
+ chroot /home/tg/build/PrawnOS/tmp.O1fT9bLDbF apt update

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
Hit:2 http://ftp.us.debian.org/debian buster InRelease
Get:3 http://ftp.us.debian.org/debian buster-updates InRelease [51.9 kB]
Get:4 http://ftp.debian.org/debian buster-backports InRelease [46.7 kB]
Get:5 http://deb.debian.org/debian bullseye InRelease [116 kB]
Get:6 http://deb.debian.org/debian unstable InRelease [146 kB]
Err:7 https://deb.prawnos.com buster InRelease
  502  Bad Gateway [IP: 172.104.217.94 443]
```

graves, 2020-10-19:
Removing flashmap and mosys from the build script still fails with:

```
+ chroot /home/tg/build/PrawnOS/tmp.O1fT9bLDbF apt install -y -d font-source-code-pro xsecurelock

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package font-source-code-pro
+ cleanup
+ set +e
+ umount -l /home/tg/build/PrawnOS/tmp.O1fT9bLDbF
+ rmdir /home/tg/build/PrawnOS/tmp.O1fT9bLDbF
+ losetup -d /dev/loop0
+ umount -l /home/tg/build/PrawnOS/tmp.O1fT9bLDbF
+ rmdir /home/tg/build/PrawnOS/tmp.O1fT9bLDbF
+ losetup -d /dev/loop0
+ rm -rf /home/tg/build/PrawnOS/PrawnOS-Shiba-armhf.img-BASE
+ echo 'FILESYSTEM BUILD FAILED'
FILESYSTEM BUILD FAILED
+ exit 1
make[1]: *** [/home/tg/build/PrawnOS/filesystem/makefile:60: /home/tg/build/PrawnOS/PrawnOS-Shiba-armhf.img-BASE] Error 1
make[1]: Leaving directory '/home/tg/build/PrawnOS'
make: *** [makefile:99: image] Error 2
```

How are successful builds being made?

rk-zero, 2020-10-19:
Well, the repo seems down ... i cannot fix that @SolidHal please have a look.

rk-zero, 2020-10-19:
As a workaround, in addition to removal of those 2 packages, remove the repo from filesystem/resources/apt/sources.list

sed will maybe fail, remove https://github.com/SolidHal/PrawnOS/blob/68a034e4d03e818073845b5666e18c99682d3ec1/scripts/BuildScripts/FilesystemScripts/buildFilesystem.sh#L219

too

SolidHal, 2020-10-20:
Yeah, sorry, the repo is down. I'm looking into it right now. I will also fix it so that the build doesn't fail if the repo is down.

SolidHal, 2020-10-20:
alright, it is back up @graves. Thanks for reporting!

SolidHal, 2020-10-20:
Currently, the packages are built locally and then copied into the apt cache during the filesystem build process so it should be able to find the packages when `deb.prawnos.com` is down. 
Out of curiosity @graves, are there `.deb` files in each of the `<repo>/filesystem/packages/<package-name>` folders?

graves, 2020-10-21:
There are, I no longer have the logs from the build failures but from what I remember it fails when trying to find the debs to copy into the apt cache.

SolidHal, 2020-10-24:
Sorry, broke this again transition to bullseye.

graves, 2020-10-26:
Yeah, that's what I ended up doing and modified the build script to add Sway window manager, running on Weston with the Panfrost drivers. Working great so far thank you for all the hard work.

SolidHal, 2020-10-28:
I think this issue can now be closed, since PrawnOS has jumped to bullseye and the repo is fully up.

closed 2020-10-28 by SolidHal