SolidEva/PrawnOS #225: arm64 gru-kevin cursor jumping and lag

SolidHal, 2020-10-25:
Symptom: 
cursor jumps large distances when input starts
"too slow" complaints from Atmal MaXTough, cros_ec, whatever mouse driver can be found in:
- /var/log/Xorg.0.lob
- journalctl

Tests: 
- occurs with panfrost on AND with panfrost disabled in the kernel in:
    - xfce
    - gnome 
    - gnome wayland
    - sway

- libinput-debug behaves properly and responds quickly to input without lag
- running glxgears or glxheads makes the symptom mostly better, jumping is much less severe but some "too slow" messages still occur
- based on that, set cpu to "performance" governor and gpu to max operating point with opp points kernel hack-- no improvement
- enabled fps debug for mesa, observed jumping issue goes away when fps of DE is ~>30 but like glxgears, glxheads some "too slow" messages still occur

So, based on that DE, xorg/wayland, input device driver, panfrost are all ruled out. 

rockchipdrm may be to blame?

SolidHal, 2020-10-27:
Tested: 
- upstream kernel 5.4.72 with panfrost and with cdn_dp blob-- no improvement
- upstream kernel 5.4.72 without panfrost and with cdn_dp blob-- no improvement
- tried another gru-kevin users kernel 5.9.1 config-- no improvement

Further, the "too slow" messages may not directly correlate to the symptom. 

The only consistent behavior I have seen, is that the mouse becomes smooth and quickly reacts to input again when the display is being rapidly updated. Examples include: 
- glxgears/glxheads
- youtube video
- htop

SolidHal, 2020-10-27:
Tried unplugging the trackpad, keyboard, and touchscreen. Can still recreate issue with just a USB mouse.

gregordinary, 2020-10-28:
Testing with gru-kevin, can confirm this behavior with built-in touchpad and external mouse. Same messages found in /var/log/Xorg.0.log. Testing with XFCE environment.

I did find a couple posts on other forums noting this behavior, both recent, and on the libinput 1.16.x series. Libinput version on fresh PrawnOS install is 1.16.2.

- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=968093
- https://forums.gentoo.org/viewtopic-t-1117556-start-0.html

gregordinary, 2020-10-28:
The warning messages for lag were introduced in libinput 1.16. 

> libinput now monitors timestamps of the events vs the current time when
> libinput_dispatch() is called by the compositor. Where the difference
> *may* result in issues, a (rate-limited) warning is printed to the log.
> So you may see messages popping up in the form of
>   "event processing lagging behind by XYZms, your system is too slow"
> This is a warning only and has no immediate effect. Previously we would only
> notice (and warn about) this when it affected an internal timer.

The threshold for the warning is 10ms:
> evdev: warn if our event processing lags by 10ms or more

Source: https://lists.freedesktop.org/archives/wayland-devel/2020-July/041529.html

This might be more compositor related than mouse/libinput related?

SolidHal, 2020-10-28:
@gregordinary do you just see the messages in the log, or do you also experience the cursor jumps if you suddenly move the mouse quickly?

gregordinary, 2020-10-28:
@SolidHal Just re-tested, my observations are:
- There is a slight lag on initial movement of the mouse, either on first use after boot, or use after 20-30 seconds of non-use. 
- If I quickly **drag** the mouse across the screen on initial use, it sort of jumps (animation not smooth) but it jumps the expected amount, relative to the distance I dragged the mouse. 
- If I **nudge** the mouse or do a short swipe on the touchpad, it seems to jump. 
- If I nudge the mouse / do a short swipe on the touchpad at ~1 second intervals, it also jumps. Seems to be about the same distance each time. 
- Continually moving the mouse fast or slow seems to behave as expected. No lag or jumping. 

Let me know if you want me to test something more specific or if you'd like log output.

SolidHal, 2020-10-28:
@gregordinary thats perfect, you described the tests I had been doing myself. 
I was starting to worry my hardware had a defect, so I'm glad you are seeing the issue as well.

mps on the rockchip-linux irc doesn't see this issue at all, I tried his kernel config but still saw this issue.

SolidHal, 2020-12-11:
Created a video demonstrating the issue.

kernel 5.9.12
mesa 20.2.3-1 from debian bullseye
prawnOS 1.3.1
arm64, gru-kevin, rk3399, T860

Used some Mesa debug environment vars in `/etc/profile` to get debug info on the running DE (gnome in this case)
The debug vars are:
```
export GALLIUM_HUD=fps,primitives-generated,samples-passed,frametime,cpu
export GALLIUM_HUD_SCALE=2
export GALLIUM_HUB_PERIOD=0
```
these and more can be found here: https://docs.mesa3d.org/envvars.html

Explanation of the tests recorded in the video:

1) I show the cursor jumps when quick mouse or trackpad movements are made. There is a noticeable delay from when I move the mouse/trackpad and when the cursor catches up. Note how the graphs barely update, especially the fps one. 
2) I show the cursor is rendered smoothly when continuous input is made
3) Moving between settings options, or triggering other graphical updates also triggers the graphs to "catch up" and update. The fps graph especially suddenly spikes
4) I start glxgears and the fps gets pinned to 60 as expected. The cursor movement is much more responsive and quick movements don't cause cursor jumps. 

https://halemmerich.com/files/prawnos-gru-kevin-cursor-lag-demo.mp4

SolidHal, 2020-12-11:
Learned that it is expected that cursor movement *not* effect the GALLIUM_HUD graphs because cursor planes bypass the gpu. 

Ruled out a bunch of additional things:

Cursor planes: 
- mod+shift+space c disables the cursor planes in weston
- simply start weston from a tty
- can confirm cursor planes are disabled by enabling the GALLIUM_HUD. When cursor planes are disabled, cursor movement will update the graphs. 
- and can also confirm cursor jumping still occurs

Mesa:
- MESA_GL_VERISON_OVERRIDE=1.0 MESA_GLES_VERISON_OVERRIDE=1.0 MESA_ES_VERISON_OVERRIDE=1.0 startx /usr/bin/i3
- cursor jumping still occurs

Tested the follwing in a tty because "rk doesn't have a text mode so anything that's actually strictly display should be reproducible at fbcon instead of x/wayland":
```
# disable other sources of display updates updates
echo 0 | sudo tee /sys/devices/virtual/graphics/fbcon/cursor_blink
# dump output from mice, try to detect lag
hexdump /dev/input/mice
# in another tmux pane, try to simulate constant video content
hexdump /dev/urandom

it seemed to be perfectly responsive with and without the source of constant display updates.

SolidHal, 2020-12-11:
Took a weston timeline of the issue along with another video. 
I took the timeline by doing the following:
1) start weston with debug flag from a tty: `weston --debug`
2) take a timeline:  `weston-debug -o <filename> timeline`
I then used https://github.com/ppaalanen/wesgr to turn it into graphs. 

During the timeline recording I did the following:
1) show cursor jumping when the input device is moved quickly and then stopped. This occurs from 2000ms to 15000ms
2) no input. This occurs from 15000ms to 17000ms
3) show the smooth cursor when input is continuously given. This occurs from 17000 to 29000
4) show smooth quick input while display is forced to high fps/constant refresh by using `hexdump /dev/urandom` in a terminal. This occurs from 35000 to 50500

Included the raw json, as well as some graphs of the different time periods: 
[weston-timeline-cursor-jump.zip](https://github.com/SolidHal/PrawnOS/files/5681543/weston-timeline-cursor-jump.zip)

and video: https://halemmerich.com/files/weston-cursor-jump-gru-kevin.mp4

SolidHal, 2020-12-11:
I did the same method described above on a veyron-speedy with panfrost. The differences are quite interesting, but first here are the files:

[veyron-speedy-weston-timeline-cursor-comparison.zip](https://github.com/SolidHal/PrawnOS/files/5682091/veyron-speedy-weston-timeline-cursor-comparison.zip)

SolidHal, 2020-12-12:
Now for the analysis. Lets focus on the first section of both graphs, the portion of the test with "cursor flicks" that cause this issue on gru-kevin. 

Lets start with correct behavior. On veyron-speedy, this section looks like: 

[image]

We can see there are multiple vblanks per "flick". Lets Look closer:

[image]

There are actually around 8 vblanks per flick. If we look even closer: 

[image]

We can see the white gap between the grey "idle" and the "delay before repaint" is around 2ms or less. 

Lets compare that with Kevin. On kevin, a single flick looks like this: 
[image]

We can see that while on speedy there were 8 vblanks, on kevin there is only 1! That explains part of the issue, around 7 frames are missing. In addition, the gap between idle and the start of the "delay before repaint" is huge on kevin, at around 110 ms. 

Looking at all of kevins graphs we can see this ~110ms "startup lag" that is nowhere to be seen on speedy. 

They grey and white backgrounds don't have a legend, but looking at the `wesgr` code I can see that the grey background means `not looping` which I guess means the white background is `looping`. My best guess here is that the loop refered to here is the weson event loop.

SolidHal, 2020-12-23:
Using perf, I was able to narrow this down to the analogix_dp drm driver, which is responsible. 

Here are some shots of the perf. Here is what the tree looks like when I move the cursor continuously:
[image]

And here is what it looks like when the cursor is moved intermittently: 

[image]

we get see that a huge amount of time in the cursor jump is taken up by a `analogix_dp_transfer`, which when annotated shows 44.51% is just a `dmb oshld`

this aarch64 ISA is documented here: https://developer.arm.com/documentation/dui0802/b/A32-and-T32-Instructions/DMB--DSB--and-ISB

so what are we transferring?

If we go up the stack a bit, we see `analogix_dp_send_psr_spd`, which then spends most of its time in `analogix_dp_get_psr_status`. So it seems like we wait some time trying to get the psr status...

PSR is Panel Self Refresh. A good description of PSR can be found here https://www.anandtech.com/show/7208/understanding-panel-self-refresh

In the trace, we for some reason can't see the direct caller of `analogix_dp_send_psr_spd`. That said, the only places that call the `analogix_dp_send_psr_spd` are 

```
static int analogix_dp_enable_psr(struct analogix_dp_device *dp)
```
and
```
static int analogix_dp_disable_psr(struct analogix_dp_device *dp)
```

we can figure out which of these is actually being called by looking up the call stack at `analogix_dp_bridge_atomic_enable`

a quick look at the function:
```
static void
analogix_dp_bridge_atomic_enable(struct drm_bridge *bridge,
				 struct drm_bridge_state *old_bridge_state)
{
	struct drm_atomic_state *old_state = old_bridge_state->base.state;
	struct analogix_dp_device *dp = bridge->driver_private;
	struct drm_crtc *crtc;
	struct drm_crtc_state *old_crtc_state;
	int timeout_loop = 0;
	int ret;

	crtc = analogix_dp_get_new_crtc(dp, old_state);
	if (!crtc)
		return;

	old_crtc_state = drm_atomic_get_old_crtc_state(old_state, crtc);
	/* Not a full enable, just disable PSR and continue */
	if (old_crtc_state && old_crtc_state->self_refresh_active) {
		ret = analogix_dp_disable_psr(dp);
		if (ret)
			DRM_ERROR("Failed to disable psr %d\n", ret);
		return;
	}

	if (dp->dpms_mode == DRM_MODE_DPMS_ON)
		return;

	while (timeout_loop < MAX_PLL_LOCK_LOOP) {
		if (analogix_dp_set_bridge(dp) == 0) {
			dp->dpms_mode = DRM_MODE_DPMS_ON;
			return;
		}
		dev_err(dp->dev, "failed to set bridge, retry: %d\n",
			timeout_loop);
		timeout_loop++;
		usleep_range(10, 11);
	}
	dev_err(dp->dev, "too many times retry set bridge, give it up\n");
}
```

shows `analogix_dp_disable_psr` is the missing link. 

interestingly, we can see that enable and disable both have their own idea about how to read psr status: 

```
	ret = drm_dp_dpcd_readb(&dp->aux, DP_PSR_STATUS, &sink);
	if (ret != 1) {
		DRM_DEV_ERROR(dp->dev, "Failed to read psr status %d\n", ret);
		return ret;
	} else if (sink == DP_PSR_SINK_INACTIVE) {
		DRM_DEV_ERROR(dp->dev, "sink inactive, skip disable psr");
		return 0;
	}
```

vs 
```
static ssize_t analogix_dp_get_psr_status(struct analogix_dp_device *dp)
{
	ssize_t val;
	u8 status;

	val = drm_dp_dpcd_readb(&dp->aux, DP_PSR_STATUS, &status);
	if (val < 0) {
		dev_err(dp->dev, "PSR_STATUS read failed ret=%zd", val);
		return val;
	}
	return status;
}
```

but they are veryyyy similar. Lets look at how how `analogix_dp_get_psr_status` is called:

```
	ret = readx_poll_timeout(analogix_dp_get_psr_status, dp, psr_status,
		psr_status >= 0 &&
		((vsc->db[1] && psr_status == DP_PSR_SINK_ACTIVE_RFB) ||
		(!vsc->db[1] && psr_status == DP_PSR_SINK_INACTIVE)), 1500,
		DP_TIMEOUT_PSR_LOOP_MS * 1000);
	if (ret) {
		dev_warn(dp->dev, "Failed to apply PSR %d\n", ret);
		return ret;
	}
```

seems like earlier we were just checking the current status, and now we are waiting for the hardware to confirm the psr status was set properly. Lets look at how `readx_poll_timeout` is defined: 

```
/**
 * readx_poll_timeout - Periodically poll an address until a condition is met or a timeout occurs
 * @op: accessor function (takes @addr as its only argument)
 * @addr: Address to poll
 * @val: Variable to read the value into
 * @cond: Break condition (usually involving @val)
 * @sleep_us: Maximum time to sleep between reads in us (0
 *            tight-loops).  Should be less than ~20ms since usleep_range
 *            is used (see Documentation/timers/timers-howto.rst).
 * @timeout_us: Timeout in us, 0 means never timeout
 *
 * Returns 0 on success and -ETIMEDOUT upon a timeout. In either
 * case, the last read value at @addr is stored in @val. Must not
 * be called from atomic context if sleep_us or timeout_us are used.
 *
 * When available, you'll probably want to use one of the specialized
 * macros defined below rather than this macro directly.
 */
#define readx_poll_timeout(op, addr, val, cond, sleep_us, timeout_us)	\
	read_poll_timeout(op, val, cond, sleep_us, timeout_us, false, addr)
```

reformatting the call to make it a little easier to look at:

```
readx_poll_timeout(
analogix_dp_get_psr_status,
dp,
psr_status,
psr_status >= 0 && ((vsc->db[1] && psr_status == DP_PSR_SINK_ACTIVE_RFB) || (!vsc->db[1] && psr_status == DP_PSR_SINK_INACTIVE)),
1500,
DP_TIMEOUT_PSR_LOOP_MS * 1000
);
```

```
#define DP_TIMEOUT_PSR_LOOP_MS			300
```

So we basically just re-run `analogix_dp_get_psr_status` until we see the status is set, and we hold up the rest of the drm functions to so. Is that really necessary??

We don't seems to do anything special on timeout beside log an error, so it seems to me like we could just set the value and move on. 

Took a look at the chromium 4.4 gru source for kevins kernel, and see that this is infact what we do. They choose to not block on PSR disable! And when that patch got upstreamed, the non-blocking part got missed. It is simply `true` where it should be `false`. 

See the referenced commit https://github.com/SolidHal/PrawnOS/commit/f0ce6513e51cb5120547c2c3c22f5472696b66c1 for the patches

commit f0ce6513 (2020-12-23): Add kernel patch for gru-kevin cursor lag, fixes #225
Added additioanl kernel patch to correct analogix_dp_transfer failue
closed 2020-12-23 by SolidHal