SolidEva/PrawnOS #226: bullseye breaks github ci

SolidHal, 2020-10-27:
```
2020-10-27T03:10:15.6424435Z I: Running command: chroot /var/cache/pbuilder/build/17524 /debootstrap/debootstrap --second-stage
2020-10-27T03:10:15.6460870Z chroot: failed to run command '/debootstrap/debootstrap': Exec format error
2020-10-27T03:10:15.6471724Z E: qemu-debootstrap failed
```

SolidHal, 2020-10-31:
Seems the fix was to add `sudo apt install -y qemu-user-static binfmt-support` *before* the docker call, as the docker-host system needs to setup the kernel correctly so that qemu-debootstrap inside the docker can then use it.

SolidHal, 2020-11-01:
aaaand the issue that we can avoid locally by building on bullseye or pulling from buster-backports gets hit: 
```
semop(1): encountered an error: Function not implemented
```
this is because the docker host (ubuntu 20.04) doesn't have a new enough version of qemu-user-static...
we would need to be on ubuntu 20.10, or we have to install a backport of the package from a ppa. Since (20.10) won't(?) be an option (see https://github.com/actions/virtual-environments/issues/1862) we will have to use a ppa for now. 
https://launchpad.net/%7Ejacob/+archive/ubuntu/virtualisation?field.series_filter=focal seems fine. 

This increases the trust risk in the build chain of these CI images. needing to use ubuntu as the host vm was already a compromise, and now we have to use a 3rd party built qemu.

SolidHal, 2020-11-01:
Unfortunately I mis-read that ppa. It only supplies qemu, not qemu-user-static.

austin987, 2020-11-15:
Well, to be fair, if you're committing stuff without making sure that CI works first, then the CI can't really help prevent breaking things.. I'm not sure what value the CI provides in that case?

SolidHal, 2020-11-15:
You are right @austin987, currently the CI is absolutely useless. The issue is at the core of github actions, https://github.com/actions/virtual-environments/issues/1862. I would need to package a new version of qemu-user-static for ubuntu 20.04 to get CI working again.
I'm not going to prevent PrawnOS from moving to bullseye just because github CI kinda sucks and refuses to provide an up to date build environment.

austin987, 2020-11-16:
Yeah, the problem is I'm not aware of any free CI that provides debian _vms_. With a container, it's trivial, but most build vms are ubuntu. Since this project strictly requires a vm (afaict), I'm not sure how to solve it.

SolidHal, 2020-11-16:
I'm backporting the ubuntu 20.10 qemu-user-static package to 20.04, should take care of it. 
I'm very surprised github doesn't keep up with the ubuntu releases.

SolidHal, 2020-11-18:
Second issue, only qemu 5.0 is available on 20.10, but 5.1 is required for armhf to avoid https://bugs.launchpad.net/ubuntu/+source/qemu/+bug/1897854

SolidHal, 2020-11-18:
looks like 5bfe2115590b47496a91a26b62f3331b45ff993c and 63ddc9abb06af20a847769db2bb2ffad05c39489 fixed it! Should float us for now.

commit 07f8bf3c (2020-10-27): try to fix #226
closed 2020-11-18 by SolidHal