SolidEva/PrawnOS #228: Pre-built images (v1.1) not booting

gregordinary, 2020-10-29:
Just downloaded both the armhf and arm64 pre-built images from the 1.1 release and tried them on the C201 and gru-kevin Chromebooks, respectively. Neither are booting from the microSD card, I get the internal speaker sound when I press CTRL+U on both Chromebooks. 

I did just build an image from source yesterday when most/all of the changes in this release were already committed and that booted fine on gru-kevin.

tehbra1n, 2020-10-29:
I have the same issue using the armhf image on my C201P.  Previous images worked fine.  I'm going to build from source for kicks on my deb vm and see what happens.

gregordinary, 2020-10-29:
Tried with a fresh build from source and that seems to be working fine, so perhaps something with the automated build process.

tehbra1n, 2020-10-30:
> Tried with a fresh build from source and that seems to be working fine, so perhaps something with the automated build process.

Same.

SolidHal, 2020-10-31:
just reuploaded the images, sorry for the issue.

closed 2020-10-31 by SolidHal