SolidEva/PrawnOS #229: Add BRCM4354 kernel module

joshbowyer, 2020-10-31:
The Veyron Minnie (C100P) comes with the BRCM4354 chip, but the module is not present in the kernel. I think it would be ideal to include the module in the kernel, but blacklist it by default as the driver is nonfree, which would allow it to be enabled if the user wishes.

joshbowyer, 2020-10-31:
The module/flag should be BCM4354

gregordinary, 2020-10-31:
This would be convenient, but also conflict with issue #28 if that's still a project goal. 

Including the firmware, even disabled, would be a blocker for the project getting FSF certification.

rk-zero, 2020-10-31:
I was conflicted on that issue before but i would now vote to NOT include non-free drivers. There is already a fork with non-free support for built-in wifi here: https://github.com/Maccraft123/ShrimpOS

Getting endorsed by the FSF would make the project waay more visible. As the c100p is discontinued but still around in US schools this would savelots of e-waste and offer a cheap solution for students to use theirs as a free computer.

joshbowyer, 2020-10-31:
Thank you for making me aware of that fork, will close this issue.

closed 2020-10-31 by joshbowyer

joshbowyer, 2020-10-31:
@SolidHal @rk-zero I know this isnt the right place for this but there is no Issues tracker on ShrimpOS and I see you guys have contributed there actively, is it possible to add an issue tracker? The latest ShrimpOS release wont boot on the C100P...

rk-zero, 2020-10-31:
I advocated a 2 kernel solution for shrimpos to @maccraft123 that does nothing more than adding a second kernel partition.

You can then switch the kernel by changing the priority of the partition with cgpt. See original chromeos partition layouts on how to do it.

This is the easiest way to have a second non-free fork that will stay usable, but im not the author of ShrimpOS ¯\_(ツ)_/¯

SolidHal, 2020-10-31:
@joshbowyer I have not contributed personally to ShrimpOS and am not affiliated in any way. ShrimpOS is a fork of PrawnOs, which is why I show up as a contributor. You will have to contact the maintainer, @Maccraft123 if you would like support.

austin987, 2020-11-24:
FWIW, I recently rebased the wifi firmware on the blobby kernel work (5.9.8) in PrawnOS here:
https://github.com/austin987/PrawnOS/tree/wifi

I've only tested on C201P, I don't have access to any other hardware, but that should be enough to start hacking on for other devices with a more recent kernel ;).

IMO though, I think a second kernel partition makes sense on its own, for safer kernel upgrades. I opened #243 for that.

Maccraft123, 2020-11-26:
I should check my email more often ._.