SolidEva/PrawnOS #230: Add ZRAM kernel module

joshbowyer, 2020-10-31:
Module or flag. This is pretty standard on ChromeOS as well as most Linux distros now (it has been added to Mobian for Pine devices too)

SolidHal, 2020-10-31:
Seems like it would give us some nice performance improvements on these low-ram systems. Could users with ChromeOS report how large of zram partitions are used on 2GB and 4GB models?

joshbowyer, 2020-10-31:
I dont use ChromeOS but on Mobian (Debian for Pine) we generally set it to 50% and all cores (4), and it works very well

SolidHal, 2020-10-31:
CONFIG_ZRAM was actually already enabled on the armhf kernel. Now enabled on the arm64 kernel as well. Next is to setup the userspace config to take advantage of it.

joshbowyer, 2020-10-31:
It isnt on my install and I installed v1.1... to enable in userspace install zramswap and edit /etc/defaults/zramswap

SolidHal, 2020-10-31:
https://github.com/SolidHal/PrawnOS/blob/f8459643606db4c4e4a489588bc7312e21e81109/kernel/resources/armhf/config#L1478 shows it is built in to the kernel. Its not a module, so it won't show up in lsmod. Actually, nothing is a module right now. on my veyron-speedy, I see `/dev/zram0` which further confirms it is enabled. 

By "on all cores (4)" are you saying there is a zram device per core?

joshbowyer, 2020-10-31:
Yes, it utilizes all cores which is optimal, otherwise you only get 1/4 the swap capacity/processing.

Ill have to double check the exact error Im getting because when I started it it screamed about the module not existing...

SolidHal, 2020-10-31:
ah, so you are talking about `cat /sys/block/zram0/max_comp_streams` not the number of zram devices to create? From https://www.kernel.org/doc/html/latest/admin-guide/blockdev/zram.html it looks like it automagically creates the right amount of compression streams

joshbowyer, 2020-10-31:
Im not sure about how it translates to the /sys blocks, but as far as the config itself those are the variables/values we found to work best. (Im not a dev, Im a linux admin who tries to help as much as he can lol, so sorry if Im missing an obvious link here)

graves, 2020-11-06:
Zram takes a parameter to set the number of devices it creates, the default is 1 which is why you only see `/dev/zram0` created.

So you'd either need to build it as a module and load it with:

`modprobe zram num_devices=4`

or add the parameter to the kernel parameters when you build the kernel.

SolidHal, 2020-11-06:
@graves thanks but my confustion is around if we want a zram device per cpu or if `max_comp_streams` is what we want per cpu. 
I'm leaning towards `max_comp_streams` since the documentation mentions it now automatically sets `max_comp_streams` based on the number of cpus