SolidEva/PrawnOS #231: PrawnOS 1.1 armhf on C201P + Gnome desktop results in crash on boot

tehbra1n, 2020-10-31:
Both using the armhf image and building from source results in a crash on boot with Gnome but not Xfce.  It seems like the boot process gets as far as starting the Gnome desktop manager, then an endless stream of error messages scrolls until the power button is held down to power of the laptop.

I'm not sure if there is an easy way to get a text copy of the output but here is a pic.  
[image]

gregordinary, 2020-10-31:
I experienced the same thing, however I was testing 4k patches on the latest 1.1 release so I didn't know if it was the patches or not. But I can confirm this happened only in gnome and not xfce

SolidHal, 2020-10-31:
Looking into it, thanks for the report

SolidHal, 2020-11-05:
kernel 5.9.4 seems to have fixed the issue, I'll make a 1.11 release soon with this fix

SolidHal, 2020-11-18:
released 1.2 which fixes this, closing.

closed 2020-11-18 by SolidHal