SolidEva/PrawnOS #236: Building in docker

dashaub, 2020-11-17:
Is building in docker not supported? I've tried unsuccessfully building  [v1.2](https://github.com/SolidHal/PrawnOS/releases/tag/v1.2) on a `buster` container with backports, and here is the output from a `bullseye` container:
```
root@c671079e2ba7:/PrawnOS# sudo make image TARGET=armhf
TARGET is armhf
make clean_image
make[1]: Entering directory '/PrawnOS'
TARGET is armhf
rm -f /PrawnOS/PrawnOS-Shiba-armhf.img
make[1]: Leaving directory '/PrawnOS'
make filesystem
make[1]: Entering directory '/PrawnOS'
TARGET is armhf
mkdir -p /PrawnOS/build/armhf
mkdir -p /PrawnOS/build/armhf/chroot-apt-cache
mkdir -p /PrawnOS/build/armhf/logs
BUILDING /PrawnOS/build/armhf/prawnos-pbuilder-armhf-base.tgz
pbuilder create --basetgz /PrawnOS/build/armhf/prawnos-pbuilder-armhf-base.tgz --configfile /PrawnOS/filesystem/resources/pbuilder/prawnos-pbuilder-armhf.rc
W: /root/.pbuilderrc does not exist
W: cgroups are not available on the host, not using them.
I: Distribution is bullseye.
I: Current time: Tue Nov 17 16:08:50 UTC 2020
I: pbuilder-time-stamp: 1605629330
I: Building the build environment
I: running qemu-debootstrap
/usr/sbin/qemu-debootstrap
I: Running command: debootstrap --arch armhf --foreign --include=apt --cache-dir=/var/cache/pbuilder/aptcache/ --variant=buildd bullseye /var/cache/pbuilder/build/18082 http://deb.debian.org/debian
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id 0146DC6D4A0B2914BDED34DB648ACFD622F3D138)
I: Retrieving Packages 
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Checking component main on http://deb.debian.org/debian...
I: Retrieving libacl1 2.2.53-8
I: Validating libacl1 2.2.53-8
I: Retrieving adduser 3.118
I: Validating adduser 3.118
I: Retrieving apt 2.1.11
I: Validating apt 2.1.11
I: Retrieving libapt-pkg6.0 2.1.11
I: Validating libapt-pkg6.0 2.1.11
I: Retrieving libattr1 1:2.4.48-5
I: Validating libattr1 1:2.4.48-5
I: Retrieving libaudit-common 1:2.8.5-3.1
I: Validating libaudit-common 1:2.8.5-3.1
I: Retrieving libaudit1 1:2.8.5-3.1
I: Validating libaudit1 1:2.8.5-3.1
I: Retrieving base-files 11
I: Validating base-files 11
I: Retrieving base-passwd 3.5.48
I: Validating base-passwd 3.5.48
I: Retrieving bash 5.1~rc2-1
I: Validating bash 5.1~rc2-1
I: Retrieving binutils 2.35.1-2
I: Validating binutils 2.35.1-2
I: Retrieving binutils-arm-linux-gnueabihf 2.35.1-2
I: Validating binutils-arm-linux-gnueabihf 2.35.1-2
I: Retrieving binutils-common 2.35.1-2
I: Validating binutils-common 2.35.1-2
I: Retrieving libbinutils 2.35.1-2
I: Validating libbinutils 2.35.1-2
I: Retrieving libctf-nobfd0 2.35.1-2
I: Validating libctf-nobfd0 2.35.1-2
I: Retrieving libctf0 2.35.1-2
I: Validating libctf0 2.35.1-2
I: Retrieving build-essential 12.8
I: Validating build-essential 12.8
I: Retrieving bzip2 1.0.8-4
I: Validating bzip2 1.0.8-4
[... 611 lines cut ...]
I: Configuring libstdc++-10-dev:armhf...
I: Configuring binutils...
I: Configuring perl...
I: Configuring gcc-10...
I: Configuring libdpkg-perl...
I: Configuring g++-10...
I: Configuring gcc...
I: Configuring dpkg-dev...
I: Configuring g++...
I: Configuring build-essential...
I: Configuring libc-bin...
I: Base system installed successfully.
I: debootstrap finished
I: copying local configuration
W: No local /etc/mailname to copy, relying on /var/cache/pbuilder/build/18082/etc/mailname to be correct
I: Installing apt-lines
I: Refreshing the base.tgz 
I: upgrading packages
I: mounting /proc filesystem
mount: /var/cache/pbuilder/build/18082/proc: permission denied.
W: Aborting with an error
make[1]: *** [/PrawnOS/filesystem/makefile:60: /PrawnOS/build/armhf/prawnos-pbuilder-armhf-base.tgz] Error 32
make[1]: Leaving directory '/PrawnOS'
make: *** [makefile:100: image] Error 2
```

SolidHal, 2020-11-17:
Docker is supported, but is more work to get functional than just a bullseye vm. 
make sure you have the following args in your docker run command:
` --privileged -v/dev:/dev`
docker is dependent on the host to setup quite a few things, including qemu-user-static. If your host has too old of a version, you will get a sepmop error later on in the build.

dashaub, 2020-11-18:
Makes sense. One more thing: adding checksums and/or signatures to the built images would be nice.

Thanks for all your work making this great project.

closed 2020-11-18 by dashaub

spoelstraethan, 2021-01-13:
I've also discovered that Balena.io has some Docker images that include qemu-user-static and have an option to register different containers as a provider for the binaries for different architectures, so you aren't necessarily limited to what is available/maintained for your host OS either (like LTS versions not having a newer version with more patches etc).