SolidEva/PrawnOS #239: installing Xfce desktop environment fails (tries to dowgrade mesa)

austin987, 2020-11-21:
I just built a new image on my desktop to test a PR. The build goes fine, but when I attempt to install, it bails out when installing mesa. Terminal output:

```
The following packages will be DOWNGRADED:
  libegl-mesa0 libegl1-mesa libgbm1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 libosmesa6
0 upgraded, 0 newly installed, 7 downgraded, 0 to remove and 0 not upgraded
E: Packages were downgraded and -y was used without --allow-downgrades
```

Installing without a desktop works fine (didn't try gnome/lxqt). I also tried adding `--allow-downgrades` here:
https://github.com/SolidHal/PrawnOS/blob/master/scripts/InstallScripts/InstallPackages.sh#L81

which worked, but I'm not sure if that's the proper fix (I assume there's probably a missing package pin somewhere).

rk-zero, 2020-11-24:
had the same issue, you can workaround that for now by adding --allow-downgrades to all 6 lines.

SolidHal, 2020-11-28:
yeah, I figured this might happen.
We should default to not downgrading with mesa. The only reason to have PrawnOS mesa builds is when debian hasn't packaged a fix that we need yet. Now that debian actually has a newer version, we should just use that. 

I'll disable the mesa build process for now since it won't be used, but will likely come in handy in the future. 

Still, this would just happen again next time PrawnOS builds its own mesa package so it will need to be worked around.

SolidHal, 2020-11-28:
this should be fixed now, reopen if needed

closed 2020-11-28 by SolidHal