SolidEva/PrawnOS #243: Add a second kernel partition

austin987, 2020-11-24:
As mentioned in https://github.com/SolidHal/PrawnOS/issues/229#issuecomment-719968121

Non-free firmware aside, I think there is solid value in providing a second kernel partition (i.e., so that the kernel can be upgraded and have the ability to revert if things break).

Once there is a second partition, https://github.com/SolidHal/PrawnOS/blob/master/scripts/InstallScripts/UpgradeKernel.sh should then be updated to write the kernel to the non-active partition (maybe with a flag to override that), and then use cgpt to update the priority of the other partition.

SolidHal, 2020-11-28:
I definitely see some value in this, especially on the armhf devices that will likely always use depthcharge. On arm64 devices, whatever bios/bootloader replacement ends up in use would have to have support for multiple kernel partitions for this to still be useful. 
A backup kernel is always helpful when upgrading.

SolidHal, 2020-12-02:
on https://github.com/SolidHal/PrawnOS/pull/249 @rk-zero 
> Thanks so much for working on this. I use a c100pa on the couch that i would love to run without a wifi dongle.
> 
> Depthcharge would allow for 2 kernel partitions and switching between them by modifying the priority via cgpt - can you mod this pull request so we get 2 kernel partitions?
> 
> Thanks again, R

I understand this is a use case many folks deal with. I'm not completely clear on how being able to boot between the two is a large convenience. By booting the blobby kernel, you have already accepted that the associated filesystem/userspace has been used with the blobby kernel. In my mind, it is better to either accept the wifi blob in your kernel and just use that or find a way to cope with a wifi dongle in the name of no blobs. I'm not judging either way. Even with just the wifi blob, these machines are amazingly libre.

Maccraft123, 2020-12-14:
There are some notes about this on ArchWiki:
https://wiki.archlinux.org/index.php/Talk:Chrome_OS_devices#cgpt_command

SolidHal, 2020-12-24:
from @Maccraft123 s Cadmium: https://github.com/Maccraft123/Cadmium/blob/master/build-all#L44-L46
this also needs cmdline change, to set root parameter to: root=PARTUUID=%U/PARTNROFF=2

also, need some way to track which kernels from which packages are in which partitions, since uname -r isn't precise enough to tell us the full package version.

SolidHal, 2020-12-24:
Could update the partitions labels as well when using cgpt to change the priority, put the kernel image package version in the label

thanks to @Maccraft123 for suggestion partition labels :D