SolidEva/PrawnOS #245: Wake from suspend breaks

joshbowyer, 2020-11-26:
When waking from suspend on the C100, everything freezes, and if i keep trying I can occasionally get it to eventually let me change tty. But when I am able to do so I cant log in as its too busy vomiting messages on the screen about panfrost and I think memory errors. This is happening with both lightlocker and gnome-screensaver.

joshbowyer, 2020-11-26:
So testing just now, when I suspend via lid close (no lock) and then reopen, the screen appears just as I left it, but nothing is clickable. dmesg shows the cpu power down and then power up, followed by:

cros-ec-spi: spi0.0: EC failed to respond in time
cros-ec-spi: spi0.0: Command xfer error (err:-110)
elants_i2c 3-0010: Unknown packet ff ff ff ff
elan_i2c 4-0015: invalid report id data (ff)

joshbowyer, 2020-11-26:
By the way I should mention that this has been the case on both PrawnOS latest image, and now also on @austin987 s fork as well, however while I wasnt able to reproduce it this morning, I only recall seeing the panfrost data dump on his fork, although it may have also been a fluke. But this shows that suspend is broken on both kernel versions, and to my memory it was also broken on debian back in 2016 (fun fact, I was the second person to have linux on a C100; the french guy who did most of the kernel/driver work helped me directly), and then also with Kali Linux. 

All that rambling to say; this has always been a problem and appears to be with the mainline kernel. I wonder if the ChromeOS kernel does something differently?

SolidHal, 2020-11-28:
if panfrost is to blame, the chromeos kernel would behave differently. 
What kernel version are you seeing this with? What PrawnOS git sha or release version? 

Those are likely red herring errors, as I have seen them all over the kernel logs even when things behave properly.

joshbowyer, 2020-11-28:
I have seen this issue with Shiba v1.2 in the releases page, and also with Austin's fork. Both behaved this way on vanilla flash.

SolidHal, 2020-11-28:
have you tried the default xfce or gnome setups from the PrawnOS installation? lightlocker has known issues with armhf platforms, so for xfce PrawnOS installs xsecurelock instead. 

What are you using as your boot device?

joshbowyer, 2020-11-28:
Installed on emmc, and this was again with all install defaults, and also tried with gnome-screensaver

joshbowyer, 2020-11-28:
and sorry this was default xfce setup

Maccraft123, 2020-11-28:
Can you run something like `dmesg -w > log` and post it on [bpa.st](url)?

joshbowyer, 2020-11-29:
Working on getting a good test for that output now, but what Ive found is that suspend works (with lid open), and gnome-screensaver works, but its when I close the lid with suspend on lid close enabled that it suspends but doesnt lock, and then wakes to a borked xfce4-session

albsod, 2020-12-05:
On Panfrost + Gnome + Xorg, colours get messed up after resuming from suspend. This doesn't happen in the Wayland session. Perhaps the underlying problem is the same?

SolidHal, 2020-12-12:
@albsod that is actually a different issue that I *think* is due to panfrost. I have observed it on both wayland gnome and xorg gnome on the c201, but not on gru-kevin.

SolidHal, 2020-12-12:
regarding this issue, I don't have a great way to debug this myself as I don't have the hardware. It would be helpful to have logs dmesg and Xorg/journalctl session logs on a fresh xfce and gnome install of prawnos. I recall certain setting in xfce making the suspend/lock behavior break, but they should be set properly now by default.

joshbowyer, 2020-12-13:
https://bugzilla.gnome.org/show_bug.cgi?id=753678

It seems this is a DPMS issue.

joshbowyer, 2020-12-13:
So, Im closing this. No idea what or how, but after trying some thing I finally found that xflock4 had no screensaver command set in xfconf. Once I set the command there the behavior is now as expected. I have no idea why it failed to autoset it as xflock4 is supposed to (it uses a for loop to see which command to use but doesnt actually probe them) but this issue is no longer an issue for me.

closed 2020-12-13 by joshbowyer

SolidHal, 2020-12-15:
@joshbowyer glad to hear it is no longer an issue. Just to confirm, is there a code change we should get in PrawnOS to avoid future pain here?

joshbowyer, 2020-12-15:
A screensaver should be installed by default and the xfconf variable for its command should be set. In my case I installed gnome-screensaver ans set the LockCommand variable to gnome-screensaver-command --lock

Without this set, the X session will freeze with "suspend on lid close" and "lock screen when going to sleep" are set

SolidHal, 2020-12-16:
xsecurelock is installed as the lock command, so that *shouldn't* be the case.

SolidHal, 2020-12-16:
tested on a c201 with PrawnOS v1.3.1, the xfconf setting is indeed set to xsecurelock. Also tested that locking the screen, and suspending manually and with lid close behave properly and activate xsecurelock.
 
@joshbowyer if you get a chance, please test with a newer PrawnOS release. I would like to ensure this is not a c100p specific bug.