SolidEva/PrawnOS #254: Gnome Settings: About shows that LLVM graphics is being used (Panfrost + Wayland)

albsod, 2020-12-05:
Under Wayland, Gnome Settings: About shows this graphics string:  "llvmpipe (LLVM 11.0.0, 128 bits) / Unknown Graphics Controller". Under Xorg it correctly shows that Panfrost is being used, however.

```
glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Panfrost (0xffffffff)
    Device: Mali T760 (Panfrost) (0xffffffff)
    Version: 20.2.3
    Accelerated: yes
    Video memory: 4014MB
    Unified memory: yes
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: Panfrost
OpenGL renderer string: Mali T760 (Panfrost)
OpenGL version string: 2.1 Mesa 20.2.3
OpenGL shading language version string: 1.20

OpenGL ES profile version string: OpenGL ES 3.0 Mesa 20.2.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
```

tehbra1n, 2020-12-06:
same here, FWIW.

[image]

Maccraft123, 2020-12-06:
Can you try with having(one at a time):
- Your user in 'video' group
- /dev/dri/* set to 0666 mode
- Launched Wayland compositor in tty1
And show what has LLVMpipe renderer and what makes use of Panfrost?

albsod, 2020-12-06:
> Can you try with having(one at a time):
> 
>     * Your user in 'video' group

My user was already in the video group

>     * /dev/dri/* set to 0666 mode

This made no difference

>     * Launched Wayland compositor in tty1
>       And show what has LLVMpipe renderer and what makes use of Panfrost?

I don't understand. Could you explain in more detail what I need to do here?

SolidHal, 2020-12-07:
this could explain https://github.com/SolidHal/PrawnOS/issues/225
if the compositor/wm isn't getting hw accel from panfrost, the cursor lag could be caused by the sw rendering not being able to keep up.

SolidHal, 2020-12-11:
On gru-kevin, gnome simply reports "Unknown Graphics Controller / Unknown Graphics Controller" which is probably correct, so this issue seems to only effect the c201, or maybe just armhf devices?

SolidHal, 2020-12-11:
Tested v1.1 on veyron-speedy, which has the mesa 10.2.1-100 packages built by PrawnOS instead of the upstream debian ones and kernel 5.4.57
The Graphics section in Gnome About is properly populated with `Mali T760 (Panfrost) /Mali T760 (Panfrost)`

I then upgraded on the kernel to 5.9.12, keeping everything else on the system the same and found the Graphics section to be incorrect, so this is likely a kernel bug or a bug with how mesa/the kernel interact.