SolidEva/PrawnOS #255: Kernel Panic When Attempting to Boot on gru-kevin

polarisfm, 2020-12-05:
Attempting to boot PrawnOS arm64 on gru-kevin causes the kernel to panic. I have tried this on multiple USB flash drives (both USB C and USB A with an adapter) and it causes the same issue each time.
This occurs on 1.1, 1.2, and 1.3. I really hope this isn't a silly configuration issue, I followed the instructions quite carefully.

`switch_root` fails because it fails to find `new_root` but I'm really not familiar enough with the boot process to figure out why this is happening without some level of guidance. The fact this only seems to be happening to me (at least judging my the other reported issues) makes me think I've just done something dumb, so I'm really sorry if that's the case. I'm very new to ARMv8 computing and even newer to using machines that use depthcharge.

Just to be clear, I am not running Coreboot (yet).

A photo of the error is attached below. It is mostly the same on each OS version (besides the things you would expect to be different like the kernel version).

[image]

polarisfm, 2020-12-07:
Update: Booting from a microSD card works fine and allows installation to other attached media (internal and external work well).

This issue seems to be specific to USB boot.

SolidHal, 2020-12-09:
Interesting! Thank you for the detailed report @polarisfm 
Let me look into this and get back to you. 

FYI, unless you specifically reflashed your kevin, you are already running coreboot with depthcharge :)

polarisfm, 2020-12-09:
Thanks!

Also my bad, I meant deblobbed Coreboot. From what I can tell it ships with some blobs in ARM TPM but based on my experience with another RK3399 machine they aren't hard to remove and don't significantly impact usability.

SolidHal, 2020-12-09:
This behavior seems to be the result of a race condition. The kernel seems to not be mapping USB devices quick enough, meaning when we try to get the list of connected devices using `blkid` the usb flash drive we are booting from is missing!

closed 2020-12-09 by SolidHal

SolidHal, 2020-12-09:
Created release with this fix: https://github.com/SolidHal/PrawnOS/releases/tag/v1.3.1

commit cb3245f9 (2020-12-09): Sometime when booting from USB the kernel doesn't enumerate devices fast enough, leaving us unable to find
the device we are booting from. Give the kernel some time. Fixes #255