SolidEva/PrawnOS #256: lilyterm fails to run, broken config

austin987, 2020-12-09:
Noticed we have 'lilyterm' (https://lilyterm.luna.com.tw/) installed, saw it before Xfce4 terminal when trying to test something on a fresh install. That failed with a GUI error, saying that ~/.config/lilyterm/default.conf is in the old format and needs to be updated.

Which is strange, because ~/.config/lilyterm doesn't exist..

Anyway, it's a lilyterm bug that it gives that error if the file doesn't exist, but to the end user, it's arguably a bug in PrawnOS. We could ship a config file, or remove it.

I'd vote for removing it, though I'm not sure how it's installed. I'm guessing as a suggestion/dependency of something, since `git grep lilyterm` shows no results.

SolidHal, 2020-12-11:
I agree, it must be tagging along as a rec from some package.

closed 2020-12-11 by SolidHal

austin987, 2020-12-14:
Thanks for the fix, but that only removes it from gnome, but not xfce (which also has it installed, that's how I noticed it ;) ).

SolidHal, 2020-12-15:
ah, good point. I only first noticed it in gnome so I assumed it was a gnome-specific issue.

commit 1f608bc0 (2020-12-11): remove undeisred gnome packages. Closes #256