SolidEva/PrawnOS #258: Prompt to add new users on (E)xpand w/ DE Install

gregordinary, 2020-12-12:
When choosing the (E)xpand option on InstallPrawnOS, and installing a desktop environment, there is no prompt to create a regular user. 

This makes sense for just expanding the installation, but for desktop environment installation, the option should be given to create a new user. Tested this installation method on both XFCE & GNOME, behavior the same for both, you end up with a DE and only a root user. 

For XFCE, you can login as root, for GNOME, it doesn't let you login.

SolidHal, 2020-12-12:
Thanks for point that out, seems like it was an oversight with some cleanup that occurred. Should be fixed now but don't have the time to test it today.

gregordinary, 2021-01-13:
@SolidHal - I tested this out on a fresh build  and unfortunately, it still doesn't prompt for user creation after choosing Expand followed by Gnome for a DE.