SolidEva/PrawnOS #261: v1.3.1 release doesn't boot from microSD if encryption option is selected

espoelstra, 2020-12-14:
I extracted v1.3.1 to a USB to install to a microSD and while the installation completes, upon rebooting without the USB inserted, I get an error from depthcharge when trying to Ctrl+U boot, presumably because the /boot partition is encrypted or the signed kernel it looks for isn't readable?

SolidHal, 2020-12-15:
Does this occur when encryption is not chosen? gru-bob support is still quite underdeveloped so I wouldn't be surprised if it just has issues booting from sd.

espoelstra, 2020-12-15:
It does actually boot when not using encryption, though I was still having some weird visual artifacts and lsusb and lspci weren't working so I didn't have much luck checking if a newer package made things better.

SolidHal, 2020-12-16:
hm, then something went wrong during installation. depthcharge beeping indicates the kernel isn't present at all, which encryption shouldn't effect. 
How did you test sd boot? did you simply dd a prawnos image to the sd card, or did you follow the same 1) boot from usb 2) install to sd card steps as before? 

I ask because my best guess is a bug in the script that installs from USB to sd

espoelstra, 2020-12-16:
Your suspicions are correct, I used the Chrome Recovery tool to write the img (renamed .bin so it sees it) to a USB and then booted from the USB and did the encrypted installation to the microSD. Oddly enough the same process but without encryption works so it's maybe just a missed conditional on the encrypted installation block.

espoelstra, 2020-12-20:
I need to test this again with the image that has the USB fixes, I'm wondering if there was another kernel config option that changed between arm64 and armhf for bob/gru that would have caused issues with the encrypted installation option.

SolidHal, 2020-12-23:
Shouldn't be, I boot head of master PrawnOS on gru-kevin encrypted on emmc. I'll have to pour some time into this when I get some.