SolidEva/PrawnOS #262: v1.3.1 USB wired network devices not recognized?

espoelstra, 2020-12-14:
Device: C101P gru-bob

I have a couple WIDELY compatible ASIX USB network cards that have worked with practically every OS/initrd that I've thrown at them, but though the `dmesg` output shows the devices as recognized, they don't show up when I do `ip addr` or `ip link` and oddly enough it seems my installation is missing `lsusb` or `lspci` whether I run those as my user or as root.

There also seem to be a bunch of Panfrost errors in the `dmesg` output, and so far I've only had it hang/lock on me once, but without networking I can't enable SSH to see if it was just a GUI hang or if the system is really down and do further debugging.

SolidHal, 2020-12-15:
Could you provide the information from dmesg about what the devices are recognized as? 
The output of lsusb would be helpful as well. It should be included in the default packages, but seems to have gone missing recently. Thanks for pointing that out. Since you don't have networking to grab the package for lsusb, you could always grab it from https://packages.debian.org/bullseye/usbutils along with the dependencies on another machine, move them over with usb, and use `dpkg` to install them. 

Otherwise, when the build CI finishes building the commit referenced on this issue, you can just use that instead of trying to get lsusb working on your own.

Maccraft123, 2020-12-15:
It looks to me like some usb ethernet controller drivers aren't enabled in default arm64 kconfig

espoelstra, 2020-12-15:
While attempting to try the generic linux-image-arm64 kernel I also discovered there appears to be an issue with how /etc/default/keyboard/keyboard is defined, it should be a file in /etc/default and not a folder with a file in it, the plymouth initramfs hook REALLY didn't like that, I had to move the file up a directory after renaming the folder and then deleting the folder once it was empty.

SolidHal, 2020-12-16:
> While attempting to try the generic linux-image-arm64 kernel I also discovered there appears to be an issue with how /etc/default/keyboard/keyboard is defined, it should be a file in /etc/default [...]

bah! this is due to a bug introduced by doing debian packaging. This is the offending line:
```
xkb/keyboard /etc/default/keyboard
```
which should instead be
```
xkb/keyboard /etc/default/
```

thanks for pointing that out @espoelstra

SolidHal, 2020-12-16:
you are right @Maccraft123 
I tried to make the arm64 config match armhf, but I guess I missed these items. they are now enabled. 
thanks for reporting @espoelstra

espoelstra, 2020-12-16:
I'll give it a try if my compile ever completes under ChromeOS on my Bob. I'll have to see if I can just sync over the kernel after I pull the latest changes and compile again.

SolidHal, 2020-12-16:
I fixed the CI, so an image should be available for test "soon" 
then you can just dd the kernel manually to whatever boot device you would like. 
make sure to wipe the kernel partition first on that boot device using: 
```
dd if=/dev/zero of=/dev/[boot-dev]1 bs=512 count=65536
```

espoelstra, 2020-12-20:
I tried just copying the kernel from the version I'd compiled but I'm guessing it wasn't properly signed as part of the compile process so it wouldn't boot. Since I didn't have any data on the microSD I just grabbed the image from the Github build and unpacked it and then wrote it to the microSD using Chrome Recovery Tool again and was able to boot, then I chose the Expand option to install. I hooked up several of my USB network cards and all of them seemed to be recognized properly, the one I finally ended up using was actually a GL.iNet Brume 1000W travel router, it presents a usb0 NIC via the USB-C port that powers the router if you have it plugged into a computer instead of a power brick. Part of the reason I love this one is the device can act as a wireless "extender" router and connect to hotel wireless or any other wireless connection and then allows clients to access it via a couple ethernet ports on the router or the USB-C like I'm using. I haven't done a deep dive on what chipsets they use for the wireless to see if it is blobby or blob-less, but one of the thing that somewhat lessens the risk even if it is blobby is that on a libre PrawnOS install if the WireGuard kernel extensions are installed I can VPN through the less open device without needing to worry about snooping as much.

One thing I noticed that was maybe weird, after selecting "no" to the installing a graphical environment because I typically use i3 and didn't need all the extra bloat, it didn't automatically reboot after the expansion was done, which I remembered happened after doing a GUI installation. I will probably do a fresh install once these changes are merged for a new release, and so I confirmed the post GUI installation reboot by running InstallPrawnOS again (I had rebooted manually myself to get the additional new space) and this time selecting the option to install a GUI and Xfce, it seemed to skip pretty fast past the expansion portion but still seemed to do something, but then it rebooted after all the packages were installed.

SolidHal, 2020-12-23:
@espoelstra 
>I tried just copying the kernel from the version I'd compiled but I'm guessing it wasn't properly signed as part of the compile process so it wouldn't boot

did you use the vmlinux.kpart or just the vmlinux? You should be using the vmlinux.kpart. 
You also need to make sure to blank the kernel partition first, the script /etc/prawnos/kernel/FlashKernelPartition.sh will do all of that for you. 

I think the script is setup to not reboot on expansion with no GUI, as there isn't really anything that requires a reboot.

Good to hear the original issue was fixed though. Closing.

closed 2020-12-23 by SolidHal

espoelstra, 2020-12-24:
Ahhh, good call, I'm not sure if I used the kpart. I'll see if I still have it sitting on the disk and give that a try since I think I tweaked another kernel option to see what it would do.

Looking forward to trying out PrawnOS on my other ARM systems soon.