SolidEva/PrawnOS #265: gru-kevin gnome lock screen freezes sometimes during wake events

SolidHal, 2020-12-22:
[suspend-lockscreen.log](https://github.com/SolidHal/PrawnOS/files/5727577/suspend-lockscreen.log)

SolidHal, 2020-12-23:
To recreate, let system suspend/lock while lid is open.

SolidHal, 2020-12-23:
recreated on blobby kernel with dptx blob for "sane" comparison: 
log: 
[blobby-suspend-log-gru-kevin.txt](https://github.com/SolidHal/PrawnOS/files/5737477/blobby-suspend-log-gru-kevin.txt)

SolidHal, 2020-12-23:
Issue recreation steps: 

1) Boot system, wait 60 seconds
2) manually suspend system with the lid left open
3) wait ~15 seconds
4) press a key to wake the system
5) observe backlight turn on, but no other signs of life on the display
6) wait 1-2 min
7) observe backlight turn off
8) press a key to wake system
9) observe expected lock screen

SolidHal, 2020-12-24:
spent too long trying to trace the behavior with perf. Couldn't recreate the issue while running perf. Seems to be some kind of race condition

cvmiller, 2021-02-16:
I see this occasionally as well. Quite by accident, After wake up I saw the desktop was visible, but nothing responded (no mouse, no clicks). I typed in my password (to unlock the screen) and it "unlocked" and I could use the desktop/mouse/etc.