SolidEva/PrawnOS #266: Hardware Support

chris00001, 2020-12-22:
I see you've got support for a couple devices including:

https://www.thinkpenguin.com/gnu-linux/penguin-usb-bluetooth-40-micro-adapter-tpe-usbbluv4
https://www.thinkpenguin.com/gnu-linux/penguin-wireless-n-usb-adapter-gnu-linux-tpe-n150usb

There are a lot more devices that you should be able to enable support for, but don't appear to have done so. I'll list them below.

I also just want to say that ThinkPenguin led the way on freeing the AR9271 and AR7010 chips. The sales pitch to Atheros's executives succeeded after an initial rejection for releasing a full set of code. It took two years just to get all the required signatures. ThinkPenguin then sold a number of USB wifi adapters based around one of two reference designs for the retail market. There hasn't actually been anybody else (companies I mean) contributing anything of significance or funding manufacturing runs in relation to the TPE-N150USB wifi adapter. The other two important people in making this code release happen were Adrian and Luis who were working for Atheros at the time (they're both big freedom people). ThinkPenguin's sales actually make up a decent even if still fairly small percentage of the overall market. Unfortunately the costs of designing and manufacturing products from scratch (or even partially so) are enormous and if we want to have a positive impact there needs to be more people purchasing and supporting these efforts rather than contributing to companies who do nothing for freedom and/or are defrauding people and/or merely pulling off public relations stunts. It might be hard to see, but progress is being made on some engineering fronts, and simultaneously we're seeing regressions occurring elsewhere as well (wifi). While there is a manufacturing run coming up there are new problems around the corner and not enough support from the community to fix these issues. It's kind of humorous considering that if every user of GNU/Linux bought a wifi adapter it would actually solve the problem, but as people spread there money out over many different companies there ends up being an insignificant demand for any one company to care enough about solving the problem.

https://www.thinkpenguin.com/gnu-linux/usb-4g-lte-advanced-modem-gnulinux-tpe-usb4glte
https://www.thinkpenguin.com/gnu-linux/4g-em7455-mobile-broadband-m2-card-gnulinux-tpe-em7455chip-europe-middle-east-africa
https://www.thinkpenguin.com/gnu-linux/4g-lt4120-mobile-broadband-m2-card-gnulinux-tpe-lt4120chip-north-america-usa
https://www.thinkpenguin.com/gnu-linux/penguin-usb-desktop-microphone-gnulinux
https://www.thinkpenguin.com/gnu-linux/usb-condenser-podcasting-microphone-tripod-pop-filter-gnulinux-tpe-pdmic
https://www.thinkpenguin.com/gnu-linux/penguin-usb-20-external-stereo-sound-adapter-gnu-linux-tpe-usbsound
https://www.thinkpenguin.com/gnu-linux/usb-20-vgadvihdmi-multi-display-video-adapter-tpe-usbgfxad2
https://www.thinkpenguin.com/gnu-linux/video-streaming-usb-webcam-noise-reducing-microphone-tpe-wcam1080
https://www.thinkpenguin.com/gnu-linux/usb-parallel-printer-cable-tpe-usbparal
https://www.thinkpenguin.com/gnu-linux/56k-usb-dial-modem
https://www.thinkpenguin.com/gnu-linux/penguin-usb-20-hi-speed-10100-fast-ethernet-network-adapter

rk-zero, 2020-12-22:
I miss any sources to those claims, especially how thinkpenguin would buy from atheros directly. This would make a great blog post story or sth, but this all reads like a sales pitch for thinkpenguin.

F.e. regarding the AR9271, thinkpenguin wrote in an older description to the TPE 150 their model is a rebrand of this unex dongle: https://unex.en.taiwantrade.com/product/575975 - they sell at 11$ per piece for a package of 2500 units (you have to write sales though)

Thats a 400% increase in price while you can get the base chip on an industry design for around 10$ even as a consumer on ali. I know only some americans that would blindly buy hardware just because the seller is "free" for that much more. Most will just check where to get a compatible chip delivered - and thats totally fine.

Therefore i vote to remove this post without further sources from a new zero-commit github account.

rk-zero, 2020-12-22:
@chris00001 Your last activity is on an apple cups driver about a label printer - i doubt you have any regards about "freedom" with the hardware you use and using a label printer is typical sales activity.

This is just spam advertising a reseller. Most of the products listed are already supported in PrawnOS (and any other somewhat recent distro) anyway.

rk-zero, 2020-12-22:
Account is most likely just founder of thinkpenguin.com Christopher Waid

Maccraft123, 2020-12-22:
@chris00001
I don't think anyone would buy the devices just to flip one kconfig option.
If you really want support for those devices in PrawnOS make a pull request with those devices supported.
Also no ARM chromebooks support M.2 or mini-PCIe devices.

chris00001, 2020-12-23:
@rk-zero 

Wow- nobody claimed that. Most products are manufactured under contract. What's your beef? Seriously. That doesn't change the fact that Atheros worked with ThinkPenguin on the project:

https://www.fsf.org/news/ryf-certification-thinkpenguin-usb-with-atheros-chip

"rebrand of this unex dongle: "

Seriously? You have no idea what you are talking about nor how much time went into getting a complete set of source code released. The dongle in question is based around a reference design that was manufactured under contract by unex for only a handful of companies of which was primarily ThinkPenguin and Sony. The first manufacturing run was done for ThinkPenguin and Sony. Each one of these manufacturing runs costs hundreds of thousands of dollars and TP is the only reason these adapters are still being manufactured in volume today. TP is not the only company who benefited from the source code release or for that matter sold an adapter with the AR9271 chip or AR710 chip. Other adapters with these chips have been sold by both TP and other companies. TP is the only company that has ever had a significant run of stock manufactured of the adapter in question however. Of two other companies that contracted for an adapter based around this same reference design one was Sony which had Sony branding and another was for some audio company. In fact if you ever tried to acquire a large number of adapters from unex directly you'd have been referred to ThinkPenguin. If you want to get technical though there was actually another two companies involved in manufacturing upstream of unex, but you wouldn't know that now would you?

Now there was an incident where some number of rejects with unex branding got leaked onto the market. An unscrupulous individual sold them to a company in Europe which initially intended to sell them all in one batch to another company. The later company rejected them after testing a smaller batch and the sale fell through. Then maybe a half dozen or so were sold at online retail over about a month, followed by one man buying a small quantity to sell through his online business. Those adapters should never have hit the market and most of that stock that got acquired initially by that first EU firm was destroyed. 

Now you very well may have received an adapter with some other product from some other company, but I can tell you that I know with almost a 100% certainty that it went through ThinkPenguin. Outside of Sony and that one audio company all other adapters were manufactured for TP and the sales went through TP.

"F.e. regarding the AR9271, thinkpenguin wrote in an older description to the TPE 150 their model is a rebrand of this unex dongle: "

What is wrong with you? ThinkPenguin has never written that and it's a false statement.

https://unex.en.taiwantrade.com/product/575975 - they sell at 11$ per piece for a package of 2500 units (you have to write sales though)

Do you have US $100k? None of this contradicts what I've said. Go ahead and reach out to them. I know for a fact that they won't sell you 2500 @ US $11 / each despite what this says and they will refer you to TP. They have a higher MOQ quantity for manufacturing so unless they happen to have some leftovers from another run this is really just misleading info. They do not typically even have 2500 units on hand because they don't sell them at retail. They are not a warehouse. You are probably looking at 10k units + and you can't even get that now even if you wanted to. They literally will not under any circumstance manufacture them for you short of some insane demand. I forget exactly what the numbers are which would be required, but if memory serves me right they're going to ask for you to sign a contract for something like 500,000 units / year which basically means you'd have to have 100% of the GNU/Linux market which needs a USB wifi adapter. Chances are you can't do better than 5%.

"Thats a 400% increase in price while you can get the base chip on an industry design for around 10$ even as a consumer on ali. I know only some americans that would buy hardware there because the seller is "free" and ships to them within days disregarding the markup. Most others will just check where to get a compatible chip delivered fast and/or cheap - and thats totally fine."

First off- that's entirely untrue and a malicious lie. TP operated a warehouse out of Europe and had shipping times in the 1-5 day range until very recently. The prices also include taxes so blame dumb legislation and thievery- not TP for the price you are seeing. There is a cost of doing business which you are not taking into account that drives up prices. The costs for a business who does nothing but resale might not have as much overhead, but then, if you don't care about free software why are you here? There was two years of energy spent working on getting code released, work toward getting support into GNU/Linux, and funding for various free software projects.

You can't even get an adapter with an AR9271 chipset in it any more outside of old stock, used adapters on eBay, and ThinkPenguin directly. ThinkPenguin is literally the only company still manufacturing adapters with it. If a company were to place an order with a factory to manufacture today it would have a cost increase of about a factor of 4x so you would probably be talking about an US $80 adapter at retail.

[... cut ...] . I'm not sure about M.2. One developer whom I work with that does a bit of development for ARM would probably be more knowledgeable about the details of various ARM systems on the market than I. Some of these devices utilize the USB bus on M.2/Mini "PCIE" so it's not inconceivable that there would be an ARM device with support potentially even if the SoC lacks PCIE. I'm pretty sure there are some that have M.2 4G modems although maybe I'm wrong and they're using an SoC with 4G modem integrated.

rk-zero, 2020-12-23:
More unbacked claims. Oh, FSF endorsed your dongle once (7 years ago), congrats. Yeah, maybe it was minifree selling the dongle with clear signs of rebrand. Still your dongle is a (surprise, using a label printer) rebranded unex dongle. You don't manufacture anything yourself, just sell what is already supported. No one owes you buying this shitty stuff.

I moved more $$ in hardware every a day than you do in a year (ok business kinda low for you rn) before going into OSS. Guess why? Money will just make you a bitter fool, resistant to new ideas. Its not our fault your business is failing, its you failing to move with the world. Sell better stuff! Actually contribute maybe? Doesnt have to be in code!

You can buy broken chromebooks on the cheap from your "connections" and fix them, will return you at least 200% (oh i forget, thats way to low roi for someone who is as great and freedom loving as you). Have business exp? Help people at minifree out with shipping and employees. Ask around what people are missing. Don't ask what OSS can do for you, ask what you can do for OSS.

Create a fork, submit a PR - it's that easy. If you want broader support just contrib to mainline upstream.

You don't have to take my points seriously, seems like next to none donations are coming in for your boring podcast: https://bitcoinwhoswho.com/address/1PrGgvU85gtvi7cFZXcSuZB8vvPQtMnSS8 which should speak volumes to what the world thinks about what you have to say.

chris00001, 2020-12-23:
@rk-zero 

I think I've realized why you reacted this way now. You're that crazy Romanian dude who sold the faulty adapters I was talking about above! You like tried to clone TP! Ha. I guess it didn't work out so well for you. Rather than trashing your competition maybe you could try being nice once in a while and IDK maybe you'd get funding for your own free software projects. Might help to move out of the EU as well. The EU is impoverishing its citizens. There is actually a lot of money out there now. It's not that hard to get a hold of it. Not like it use to be anyway. There have actually been millions of dollars flowing into free software projects like this from a variety of sources over the past 4-5 years. Some of them have been embarrassing wrecks and still managed to pull in significant dollars. 

"Yeah, maybe it was minifree selling the dongle with clear signs of rebrand. Still your dongle is a (surprise, using a label printer) rebranded unex dongle. You don't manufacture anything yourself, just sell what is already supported. No one owes you buying this shitty stuff"

You got me. You may be wrong about everything- but I can agree that TP sucks at branding and marketing. Maybe some TP money should be redirected from funding free software efforts and be put into proper branding. NOT! This would also conflict with your bitchy little whining about how nobody is funding your worthless ass. Your problem is your a dick.Your a dick to your customers your a dick to other people. Stop being a dick and maybe the next time there is funding available people would make an effort to reach out to you about the 10s of millions of dollars they got in free money earmarked for free software projects. What is humorous is even the other little bitch got paid and so did at least one other worthless project. If he could manage it (though unlike you the other person I'm thinking about is not a dick) you should be able to too. Just shut up once in a while and stop behaving poorly to those who could potentially help you.

"I moved more $$ in hardware every a day than you do in a year (ok business kinda low for you rn) before going into OSS. Guess why? Money will just make you a bitter fool, resistant to new ideas. Its not our fault your business is failing, its you failing to move with the world. Sell better stuff! Actually contribute maybe? Doesnt have to be in code!"

OK- so I was right. Well, you are the Romanian ass hole. Seriously- TP has been around longer, has better friendlier employees (you still working out of your mom's basement?), actually cares about free software, and is doing more than you could ever hope to do. The problems that need solving most of all right now are largely engineering problems. They don't get solved overnight. They're not entirely software projects and you're failure to realize that is whats going to result in your demise eventually. You can only clone, copy, and I would say steal, but copying isn't theft, your way to success to a certain point (not that I think you've even succeeded here). Eventually you got to pick up some new social skills and work with other people on projects you may not entirely agree with. I'm ashamed for lowering myself to your level. The mom joke was a bit much. Sorry. Will try to refrain further.

"You can buy broken chromebooks on the cheap from your "connections" and fix them, will return you at least 200% (oh i forget, thats way to low roi for someone who is as great and freedom loving as you). Have business exp? Help people at minifree out with shipping and employees. Ask around what people are missing. Don't ask what OSS can do for you, ask what you can do for OSS."

You have zero business skills. I'm not against fixing broken chromebooks entirely- I wouldn't have posted here if I thought that, but you're not going to be able to make a business out of selecting select chromebooks to sell and turn that into a means of funding serious efforts at solving low level problems. We don't need more developers. We need more funding for engineering projects so that we eventually have chips and cost effectively solutions to manufacture devices that don't fail on people after 2 years because a certain manufacturer utilizes proprietary components or software company discontinues support for said hardware.

I like LibreBoot, and I like PrawnOS, but they're not going to solve the fundamental problems without focusing on getting code released for wifi chips, SoC's, GPUs, and other components. Nor are they going to bring in the funds needed to cover the costs to that end. That doesn't mean the efforts are worthless, but they're not the only thing that matter.

If you want the prize (free software) sometimes taking the slower roundabout path (may not be entirely free even) is the difference between appearing to be doing something (hacking on code that will never be useful in an entirely free system) and actually doing something (like funding engineering efforts to solve the high costs of manufacturing devices that negatively impact the ability of our community to get free software friendly hardware manufactured since nobody can currently do so at a price where it can at least break even if done). Too many people are basically taking advantage of others to advance personal agendas (usually profit) and attacking the very people that are actually doing something.

[... cut ...]
Some people have better things to do than sit around and chat on IRC all day. Ya know- like this conversation here isn't productive. It's just one huge waste of time and that's basically what IRC, Telegram, Facebook Twitter, Mastodon, Matrix, and all these social networking ish type apps and sites have become. A big waste of time. You want to be productive and get something done? Stop typing. Go work on something actually useful. And I do mean stupid pointless angry messages on forums, social media, and wherever else you get into arguments with people on stupid shit that don't matter.

SolidHal, 2020-12-23:
Locking the issue for now while I try and sort out the myriad of claims made here. I don't appreciate the way this discussion is being handled. PrawnOS is not the place for this. 
Will revisit this following the holidays.

closed 2020-12-24 by SolidHal