SolidEva/PrawnOS #273: v1.3.1 armhf exhibits a lot of errors after expansion and installation of the GUI (Xfce)

spoelstraethan, 2021-01-01:
I did a flash of the armhf to a microSD and then enabled USB boot and rebooted and used Ctrl+U and then chose the expansion installation option and then installed the GUI but after a reboot it appeared frozen. I _was_ able to change to a VTY and stop lightdm (after a couple attempts). The error messages I saw in the console were related to devfreq and panfrost errors complaining about not being able to update frequency transition information.

spoelstraethan, 2021-01-01:
I'm currently grabbing the image from #269 to see whether the changes pushed since v1.3.1 help at all. I was going to try the blobby image but it appears that currently isn't saving the artifacts as part of the build.

austin987, 2021-01-01:
That's correct, PrawnOS is explicitly no blobs, so distributing blobby builds doesn't exactly fit with that idea :).

Try the artifacts from https://github.com/austin987/PrawnOS/actions/runs/453618559, it fixes a few different regressions I've seen in the last few days (note: I only tested with xfce).

spoelstraethan, 2021-01-02:
I'm seeing the same errors in that build, but oddly after running `systemctl disable lightdm` the next time I booted it still started, but it must have been triggered by another process and it actually worked correctly (though there still appear to be some errors happening in the background).

The possibly bigger issue with the "working" change, was that even though I chose the graphical installation it never prompted me to create a new user, so I had to log in as root in the lightdm session, this is terrible from the least privilege model recommended by security folks the world over.

closed 2022-07-26 by SolidHal