SolidEva/PrawnOS #282: Add blobby kernel to APT repo

joshbowyer, 2021-02-20:
So forgetting that I had my repo set to Prawn, I did an apt upgrade which wiped my blobby kernel (with BCM4354 wifi driver). It would be nice to have that blobby kernel with that driver built in the repo (once installed manually, thereby preserving your FSF endorsement) one could apt-mark linux-$(version) kernel package and an upgrade would only look for updates to linux-$(version)-blobby, thereby not nuking the internal wifi for those that use it. I looked through the pulls and it looks like this is mostly ready?

joshbowyer, 2021-02-20:
@austin987 tagging for visibility

austin987, 2021-02-22:
IMO it's ready, pending a decision by SolidHal. At least I'm happy to work towards it, if they're okay with it.

cracket, 2021-04-24:
Sorry for hijacking this issue discussion, but I'm trying to understand how to use PrawnOS with blobby wireless driver BCM4354 in Asus C100 (installation, configuration etc.). Is there any howto or docs which I can follow? @austin987

austin987, 2021-04-25:
I don't have an Asus C100, so I can't say (and FYI, PrawnOS doesn't support blobby wireless).

peter-marshall5, 2021-04-26:
I got the internal wifi card working on my Asus C201 with the blobby kernel by enabling the BCM4354 driver as a loadable module in the kernel config and setting it to support the SDIO bus. I then had to add the firmware files to /lib/firmware/brcm: brcmfmac4354-sdio.bin from the package firmware-linux-nonfree and brcmfmac4354-sdio.txt which I downloaded from https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/master/overlay-veyron/chromeos-base/chromeos-bsp-veyron/files/firmware/brcmfmac4354-sdio.txt

peter-marshall5, 2021-04-26:
@cracket I actually compiled the PrawnOS kernel myself by running `make BLOBBY_BUILD=1 TARGET=armhf kernel_config` to configure the kernel and then `make BLOBBY_BUILD=1 TARGET=armhf` to build PrawnOS with the blobby kernel. I also had to uncomment line 90 in scripts/BuildScripts/KernelScripts/buildKernel.sh so that it would build loadable modules (I had configured the kernel to build the wi-fi driver as a module to make debugging easier) and I had to add the line `make -C $KERNEL_BUILD ARCH=$KERNEL_ARCH INSTALL_MOD_PATH=$outmnt modules_install` after line 129 in scripts/BuildScripts/ImageScripts/installKernelToFs.sh so that it would put the modules in /lib/modules.

However, you can set the configuration to build the wifi driver directly into the kernel. It may also be possible to include the firmware directly into the kernel, similarly to the libre firmware for the HTC wifi dongles.

cracket, 2021-05-06:
@peter-marshall5 thank you for hint - it saved me a lot of work. Just curious if you were able to make bluetooth working?
I'm searching now all c100 available distributions and looks like all have issues with broadcom bluetooth.

peter-marshall5, 2021-05-06:
@cracket I'm glad I was able to help. Unfortunately, I wasn't able to get bluetooth working when I enabled the bluetooth drivers in the kernel. However I didn't play around much with this since I don't use bluetooth on my chromebook.

dnowen, 2021-05-08:
hi @peter-marshall5 can you explain further in detail how you used the blobby kernel to enable the BCM4354 driver as **_loadable module_** in the **_kernel config_** to support SDIO bus? I am a beginner with no clue how to add blobs to make internal wifi work. I tried this fork project [ShrimpOS](https://github.com/Maccraft123/ShrimpOS) (that includes the wifi blob) but it wont boot because it has two partitions.

peter-marshall5, 2021-05-10:
@dnowen To add support for the internal wifi card to PrawnOS, you will need to compile it yourself on a Debian system. This process is fairly involved and there is a chance of running into errors.

First, you will need to clone the github repository:
`git clone https://github.com/SolidHal/PrawnOS`

You will then need to enter the project directory:

`cd PrawnOS`

After that, you need to run this to configre the kernel:

`make BLOBBY_BUILD=1 TARGET=armhf kernel_config`

A text-based menu will appear with lots of submenus and options. It's sometimes easy to get lost in here. Use the up and down arrows and enter key to navigate to "Device Drivers", then "Network Device Support", then "Wireless LAN". Navigate to "Broadcom Devices" so that it's highlighted and press "y" to enable it. Then go down to "Broadcom FullMAC Driver" and press "y" to enable that. Finally, make sure that "SDIO bus interface support for FullMAC driver" is enabled (I think it is by default). Now navigate to Exit with the right arrow and press Enter to return to the previous menu repeatedly. Make sure that "Yes" is selected and press enter when asked "Do you wish to save your new configuration?" Congratulations, you have configured the kernel to support the internal wifi card.

Now you can either build the entire PrawnOS image or just compile the kernel and add it to a prebuilt image. To compile the entire image, run: `sudo make BLOBBY_BUILD=1 TARGET=armhf` and then use a disk image writer to write the resulting image (PrawnOS-Shiba-armhf.img, not PrawnOS-Shiba-armhf.img-BASE) to a USB drive.

To compile the kernel only, run: `make BLOBBY_BUILD=1 TARGET=armhf kernel`. When it finishes, write a prebuilt PrawnOS image to a USB drive. Then run `sudo dd if=build/armhf/linux-*/vmlinux.uimg of=/dev/sdX1`, substituting "/dev/sdX" with the device path for your USB drive. Note the "1" after the device path to tell dd to write to the first partition. This is most likely "/dev/sdb" but you can check by running `lsblk` to see the attached drives and their device paths. Make sure that you select the right device as the first partition will be overwritten with the kernel that you compiled.

After compiling either the entire image or just the kernel, you should be able to mount the second partition of the USB drive. Run `sudo mount /dev/sdX2 /mnt` substituting "/dev/sdX" with your USB drive again. Note the "2" after the device path to make Linux mount the second partition. Then run the following commands: `wget https://chromium.googlesource.com/chromiumos/overlays/board-overlays/+/master/overlay-veyron/chromeos-base/chromeos-bsp-veyron/files/firmware/brcmfmac4354-sdio.txt` to download the first firmware blob. Then run `sudo mkdir /mnt/lib/firmware/brcm` and `sudo cp brcmfmac4354-sdio.txt /mnt/lib/firmware/brcm/` to copy the firmware blob into PrawnOS. For the next firmware blob, it should already be on your computer if you have the package "firmware-linux-nonfree" installed and you can run `sudo cp /lib/firmware/brcm/brcmfmac4354-sdio.bin /mnt/lib/firmware/brcm/` to copy it into PrawnOS.

If you did everything correctly, you should be able to insert the USB drive into your Chrome device and press Control+U on the "OS verification is off" screen to boot into PrawnOS. Typing "ip link" should show a device named "wlan0" if the wifi card is working.

I didn't know about ShrimpOS until now. It would have saved me a lot of work early on when PrawnOS didn't have built-in support for Panfrost or the blobby kernel. You mentioned that ShrimpOS was unable to boot on your device. This might be because you have a device with an ARM64 CPU, in that case images for ARMHF devices won't boot since they use a different format. If this is the case, you will need to replace "TARGET=armhf" with "TARGET=arm64" in all of the make commands. It also could be that you forgot to enable booting unsigned kernels and booting from USB in ChromeOS after factory resetting your device.

dnowen, 2021-05-10:
> FWIW, I forked PrawnOS for precisely this reason (https://github.com/austin987/crawfishos/), and have binary builds available for c201p (I don't have access to arm64 hardware, never tried to test/build for that).

wow! I did not see this either. I will check this out. I also was referred to this [Cadmium](https://github.com/Maccraft123/Cadmium) for the C201P...however there are no releases for veyron_speedy yet. And I have no I am realizing now I cannot build these myself, I am getting tons of errors on the old Thinkpad I have with debian distro. I think because it needs to be built on an arm ? I dont know but it may be of use to someone with more knowledge than me...

peter-marshall5, 2021-05-10:
@dnowen Wow, I didn't know about Cadmium either. I'm going to try building it and running it on my Chromebook to test the graphics performance since the Panfrost driver isn't exactly stable yet.

austin987, 2021-05-10:
Never tried, I don't use bluetooth.

dnowen, 2021-05-10:
> @dnowen Wow, I didn't know about Cadmium either. I'm going to try building it and running it on my Chromebook to test the graphics performance since the Panfrost driver isn't exactly stable yet.

let me know how it goes because I cant get audio now on Crawfish either, wifi works though which is good

peter-marshall5, 2021-05-11:
I just built Cadmium successfully and it booted on my C201. I can post my image if you want it. Wifi works out of the box but I'mnot sure aout sound and Bluetooth yet.

dnowen, 2021-05-11:
> I just built Cadmium successfully and it booted on my C201. I can post my image if you want it. Wifi works out of the box but I'm not sure aout sound and Bluetooth yet.

@peter-marshall5 that would be great if you can post your image (maybe to your git?). Thank you! 
From what I gather from the developer, he said it should _all work_, he just didnt do a build release yet.

peter-marshall5, 2021-05-11:
Github wouldn't let me upload files larger than 25mb so I uploaded it to my Google drive: https://drive.google.com/file/d/1axI_FQr6F2ciA0-Euprm2_Q9VzxCMb87/view?usp=sharing

Sound and bluetooth weren't working out of the box on Cadmium but maybe you need to install to the internal storage for it to work. I got sound working on my PrawnOS install by running a script I found online one time that changes some ALSA mixer settings. I had to modify it a bit to make it target the internal sound card. I'll try to find it and link it.

dnowen, 2021-05-12:
> Sound and bluetooth weren't working out of the box on Cadmium but maybe you need to install to the internal storage for it to work. I got sound working on my PrawnOS install by running a script I [...]

Thanks, downloaded the image, ill flash it now!

yoonsikp, 2022-06-04:
New Kernel build if anybody is interested

https://github.com/yoonsikp/PrawnOS/releases/tag/v5.15.6

Wooty-B, 2022-09-11:
@yoonsikp line 39: Unknown device! Can't determine emmc devname. Please file an issue with the output of fdisk -l.

I don't have internal WiFi or an ethernet dongle and fdisk doesn't come prepackaged with PrawnOS. Please advise!

Wooty-B, 2022-09-11:
@peter-marshall5 What happens when you try and install ca-certificates-java under Cadmium? I always receive a segmentation fault breaking all Java packages. Thanks!

peter-marshall5, 2022-10-04:
[https://github.com/SolidHal/PrawnOS/pull/308](https://github.com/SolidHal/PrawnOS/pull/308)

SolidHal, 2022-10-21:
Closing this issue as I have no plans to expand "official" support for the blobby kernel. The blobby kernel is offered for users that really want to build their own images, and for developer testing, but is not supported. 

I'm happy to accept a PR to fix the issue where the blobby kernel gets updated by a non-blobby kernel package.

closed 2022-10-21 by SolidHal