SolidEva/PrawnOS #283: Failing to build on Latest bullseye

lluixhi, 2021-03-11:
Keyring isn't missing, any ideas on why it's failing?

```
I: Running command: debootstrap --arch armhf bullseye --include build-essential,ca-certificates,gnupg,init,locales,openssl --keyring=/home/adb/git/PrawnOS/filesystem/resources/apt/debian-archive-keyring.gpg /home/adb/git/PrawnOS/tmp.8EAvnCOaa2 http://ftp.us.debian.org/debian --cache-dir=/home/adb/git/PrawnOS/build/armhf/debootstrap-apt-cache/
E: No such script: --keyring=/home/adb/git/PrawnOS/filesystem/resources/apt/debian-archive-keyring.gpg
+ cleanup
+ set +e
+ umount -l /home/adb/git/PrawnOS/tmp.8EAvnCOaa2
+ rmdir /home/adb/git/PrawnOS/tmp.8EAvnCOaa2
+ losetup -d /dev/loop19
+ umount -l /home/adb/git/PrawnOS/tmp.8EAvnCOaa2
+ rmdir /home/adb/git/PrawnOS/tmp.8EAvnCOaa2
+ losetup -d /dev/loop19
+ rm -rf /home/adb/git/PrawnOS/PrawnOS-Shiba-armhf.img-BASE
+ echo 'FILESYSTEM BUILD FAILED'
FILESYSTEM BUILD FAILED
+ exit 1
make[1]: *** [/home/adb/git/PrawnOS/filesystem/makefile:70: /home/adb/git/PrawnOS/PrawnOS-Shiba-armhf.img-BASE] Error 1
make[1]: Leaving directory '/home/adb/git/PrawnOS'
make: *** [makefile:97: image] Error 2
```

austin987, 2021-03-11:
I haven't dug into it deeply, but it looks like debootstrap changed.

Try https://github.com/austin987/crawfishos/commit/d7b79c84caece4d4c3ff3a3080d82c7089cbab8a

SolidHal, 2022-07-26:
This has been fixed for a bit, closing

closed 2022-07-26 by SolidHal