SolidEva/PrawnOS #285: Use free firmware-ath9k-htc package from Debian instead of building from source

gregordinary, 2021-05-04:
Debian now has "[firmware-ath9k-htc](https://packages.debian.org/search?keywords=firmware-ath9k-htc) - firmware for AR7010 and AR9271 USB wireless adapters" in the repository for bullseye and sid. Can this be worked into the build process to save time compiling?

SolidHal, 2022-07-26:
Oh, thats super cool, definitely

mvhulten, 2023-05-06:
Yes, this would be great.

I installed PrawnOS `bullseye` on a C201 and I could not use my Atheros AR9271 Wi-Fi dongle (got it from [Technoethical](https://tehnoetic.com/)).  My solution was to save the packages `firmware-linux-free_20200122-1_all.deb` and `firmware-ath9k-htc_1.4.0-106-gc583009+dfsg1-1_all.deb` on a USB stick and then install with apt on the Chromebook.

Making this automatic (during install) would be a good usability improvement.

Thanks for PrawnOS and good documentation!