SolidEva/PrawnOS #295: Android-tethering doesn't work. Maybe missing rndis_host.

fljon, 2022-01-26:
Can't connect to the internet with tethering. Maybe the rndis_host module is missing.
Device (Fairphone3) shown with lsusb:
Bus 001 Device 008: ID 18d1:4ee1 Google Inc. Nexus/Pixel Device (MTP)
But it isn't shown as networkdevice (usb0).
Tethering works on my Thinkpad.
Also tethering works on the installationsimage PrawnOS Shiba v1.3.1.
After an Update it's missing.
It would be nice if you can readd it! I need this for traffeling!
Thanks a lot!

Sharelter, 2022-09-08:
This problem also occurred after I installed PrawnOS on my C101P machine. 
I also tried use a usb eth adapter but still not work.
The usb eth adapter only works when I plug this before I boot up.

oxirane, 2024-04-21:
Same issue on Samsung chromebook plus.

lsusb identifies the phone as a CDC data driver but the driver field is empty:

```
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 65, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 66, If 0, Class=Vendor Specific Class, Driver=asix, 480M
        |__ Port 3: Dev 67, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 3: Dev 70, If 0, Class=Miscellaneous Device, Driver=, 480M
            |__ Port 3: Dev 70, If 1, Class=CDC Data, Driver=, 480M
            |__ Port 4: Dev 68, If 0, Class=, Driver=, 480M
```

In theory this is fixable on an installed system by adding lines related to RNDIS to the loader.conf file:

```
CONFIG_USB_NET_RNDIS_HOST=m
CONFIG_USB_NET_RNDIS_WLAN=m
CONFIG_USB_F_RNDIS=m
CONFIG_USB_CONFIGFS_RNDIS=y
CONFIG_USB_ETH_RNDIS=y
CONFIG_USB_FUNCTIONFS_RNDIS=y
```

However, there is no such file for the running 5.15 kernel in /boot/. Where is the conf file?

NB: this is an install on an sd card. 

Alternatively, you could boot the OS using a different kernel that has those drivers (I have a 6.6.13 kernel installed after an update to bookworm - the above lines came from that config). I cannot work out how to change the kernel the system boots from however, as I'm not familiar with u-boot.