SolidEva/PrawnOS #298: deb.prawnos.com signature verification failed

jasonkfirth, 2022-06-05:
I just installed prawnos on my C201p and used a USB ethernet dongle to connect to the Internet. I can't update the prawnos specific packages because I get a signature verification failed message. I think perhaps expired signature? I went to the website and ran the suggested command line for updating the key, but that did not change anything (it actually said "no change")

ericfont, 2022-06-10:
Also chiming in...I installed it a few weeks ago, and `apt update` was working fine, but today I tried to do `apt update` but got the EXPKEYSIG error:
```
deb.prawnos.com bullseye InRelease
  The following signatures were invalid: EXPKEYSIG 37BB143E689B559D PrawnOS <hal@halemmerich.com>
```
The rest of the regular ftp.us.debian.org repos were accessed fine.

jjustra, 2022-06-12:
Same here .. cannot update. `apt-key list` says :
```
/etc/apt/trusted.gpg
--------------------
pub   rsa4096 2020-05-09 [SC]
      649E 4110 0BD7 1322 5E03  FA62 25E5 C54F 2EF7 935C
uid           [ unknown] PrawnOS <hal@halemmerich.com>
sub   rsa4096 2020-05-09 [E]
```

jjustra, 2022-06-15:
Hmm ... https://deb.prawnos.com/dists/bullseye/InRelease file contains ``Date: Wed, 16 Dec 2020 03:57:44 UTC`` ... I am by far not versed in these files, but maybe there is nothing to update from. And as I see now, merges halted in 2020 .. so this distro is probably not in active development anymore ?

SolidHal, 2022-07-25:
Sorry, I'm working on getting a new release out with a fix for this. For now you can just remove the repo from your `/etc/apt/sources.list` to get rid of the annoyance

SolidHal, 2022-07-26:
Alright, I published a new signing key and re-signed the available packages
You may have to re-add the public key using the instructions at https://deb.prawnos.com/ if you still see issues

jjustra, 2022-07-26:
Thanks a lot, works perfectly.

closed 2023-01-23 by SolidHal