SolidEva/PrawnOS #301: Review held kernel patches, investigate additional kernel patches

SolidHal, 2022-07-27:
see some here https://github.com/urjaman/linux-for-my-autobuilder/tree/linux-kbb-c201-stable
some mentioned in https://github.com/SolidHal/PrawnOS/pull/299/files

Maccraft123, 2022-07-27:
There's also https://github.com/Maccraft123/Cadmium/tree/master/kernel/patches

kapouer, 2022-09-02:
Hi, i've been playing a lot with my veyron/speedy c201 chromebooks (i have three of them),
up to the point it's possible to install debian/bookworm, and thanks to libreboot and depthchargectl,
it's possible to install mainline debian armmp-lpae packages and get gnome running pretty smoothly.
With three reserves:
- kernel command line must have "gpt rootwait"
- blacklist btsdio
- emmc still takes a long time to reset

For that last point, I wonder if someone is currently trying to get the floating patches about emmc reset accepted into upstream linux kernel. Is there ?

kapouer, 2022-09-02:
Well actually there's a fourth point, a PITA, sometimes wakeup from suspend won't work.