SolidEva/PrawnOS #303: Asus Chromebook C100P problem with wifi connection

realdaly, 2022-09-04:
I have c100p and I installed PrawnOS on external SD Card (the same SD Card I used to build the image to), the installation process went just fine with no errors, but after the system is done installing (graphical environment) I tried to connect to wireless network but it seems like there's no WiFi driver or something because the network window says "No network devices available".
The version I installed is the latest PrawnOS Shiba v1.3.2, Jul 28, 2022
Any solutions plz??

Sharelter, 2022-09-08:
Maybe because PrawnOS is an operation system which aims to be "pure free" , and the driver of wireless module is non-free.
PrawnOS disableed the loading of non-free firmware, if you check the output of `dmesg`, you may find message about it.
( The README of this project claimed that [C201](https://github.com/SolidHal/PrawnOS#build-the-wifi-dongle-into-the-laptop)  model need a usb wifi dongle to connect wifi without the non-free hardware that built-inside the machine. )

May be you need a usb wifi dongle which have free driver to archive the purpose of PrawnOS ?
Or if you just wanna a Linux system you can try [archlinuxarm](https://archlinuxarm.org/platforms/armv7/rockchip/asus-chromebook-flip-c100p).

Sharelter, 2022-09-08:
I use C101P and the wifi hardware is from marvell .
You can use lsusb or lspci to check if there is a hardware looks like a wifi module :)

realdaly, 2022-09-10:
Yeah I just read that after I opened this issue haha, I didn't know it was intentionally disabled at first.
As for Arch Linux, I've tried many times to install it on my Asus C100P, everything goes just as the instructions say, but when I try to boot into Arch it gets stuck at some "network configuration" and I found many people suffer from this same issue.
I tried the pre-built Kali images and they work just fine, but since these images are old I need to install new packages like python3.10 which ruins the whole system as soon as it gets installed.
I've tried to use the build scripts to generate a newer image of Kali Linux but the script doesn't seem to work.

Sharelter, 2022-09-14:
emmm...
lspci is a command utility to show your devices on the PCIE bus.
When U boot into the freshly installed archlinuxarm, and logged in. Just type `lspci` and hit enter to see the output. 
I temporarily used "usb network-sharing" with my phone to get network access before network configuration is finished.
U can also try it ~

realdaly, 2022-09-14:
Okay thank you so much🌹  I'll give it a try

cracket, 2022-09-22:
@realdaly  just to correct - C100P does not have any pci, you can use either builtin dual wifi+bt card or try to plug in external wifi via usb, so I'd rather suggest lsusb (or maybe lsmod if you want to list loaded kernel modules). 

If you still want to use builtin WiFi card you can leave PrawnOS running, and just replace kernel with modules& firmware to the one supporting non-free drivers. As far as I can see archlinuxarm does not provide working kernel for c100 wifi too

realdaly, 2022-09-27:
As I said to @Sharelter I'm a complete beginner in linux and Idk how to build images or anything, in fact I use windows but because c100p supports only linux so I'm looking for a new working image (idc if it was prawnos or anything else) 
I tried old kali images but when I run "apt update && apt upgarde" the system stops loading and I can't boot to it anymore.
Couldn't install Arch linux as well, I get stuck at some network config.
Right now I'm just looking for any linux built image so I can use my c100p with vscode and some other packages like python3.10.

cracket, 2022-10-01:
if you like PrawnOS installation, and you are running system from microsd card,  switching to blobby kernel with WiFi driver should be easy with following steps:
- obtaining kernel image (single binary file) and modules directory of blobby kernel
- overwriting partition with kernel image (make sure new image is not bigger than a partition size) with dd
- copying into /lib/modules  kernel modules which match new kernel image (mount & cp)
- copying into /lib/firmware blobs for wifi (this can be done anytime as it does not damage boooting the system) (mount&cp)
in issue #282 @peter-marshall5 provided good description how to build blobby kernel on your own.

realdaly, 2022-10-03:
@cracket thanks a lot man! I will definitely give it a try.

SolidHal, 2022-10-21:
@realdaly you can build PrawnOS with a blobby kernel if you'd like by passing `BLOBBY_BUILD=1` to `make`
though this is unsupported, so I make no promises of its functionality

closing this issue for now since its not a PrawnOS bug/issue

closed 2022-10-21 by SolidHal