SolidEva/PrawnOS #304: Instructions on Enabling Internal WiFi

Wooty-B, 2022-09-11:
I am well aware of the other posts on this, but requests for this have been ongoing for years with no interest from the maintainer. The problem is VERY few Debian installations work for the C201p platform. I am trying to get Cadmium working but a critical package is broken I need.

The problem is many of us are beginners trying to learn on ARM, and the ARM SoC arena usually has specific quirks to learn about their specific platforms. I have tried and failed at adding blobby wifi to my current install but am having no success. If the option will never be available could we at least get step-by-step instructions to help enable this?

IMPORTANT:
I attempted by adding internal WiFi to my first C201p, which had different colored wires than the diagrams posted. I reached out and never got a reply so attempted to pin this out myself and ended up frying my board.

I then ordered a refurbished board from AliExpress and it arrived with pulled traces on the mouse pad. I then bought a working unit on eBay, putting me almost $200 in the hole specifically for PrawnOS. 

I understand these were the risks however I wanted to share my experiences since there really isn't a simple Debian alternative with internal WiFi supported. I just refuse to sacrifice a usb port when I am not concerned about closed-source WiFi drivers.

If there are other active Debian alternatives besides Cadmium I am all ears. Thanks for hearing me out.

rtsn, 2022-09-11:
Hey I've had similar experiences as you and I eventually gave up on PrawnOS after finding the fork https://github.com/austin987/crawfishos 

I believe the fork is dead and I haven't dared to run apt upgrade but at least it works.

austin987, 2022-09-16:
> I believe the fork is dead and I haven't dared to run apt upgrade but at least it works.

I wouldn't say it's _dead_, but PrawnOS was dormant for a while, so there wasn't much to do.

Since PrawnOS has updated, I've been busy, and no one has offered to help, so yeah, it may take a while.

That said, apt upgrade should mostly be fine, aside from the kernel. Upgrading that will wipe out the wifi blob.

SolidHal, 2022-10-21:
@Wooty-B 
As I mentioned on a related issue, you can build prawnos with a blobby kernel yourself by adding `BLOBBY_BUILD=1` to the make command. crawfishos is another alternative if you can't/won't do that yourself.

> requests for this have been ongoing for years with no interest from the maintainer

Its not a lack of interest, it would be counterproductive to  the goals of this project. PrawnOS is intended to be as blob free as possible. Publishing a release with blobby wifi would defeat that purpose. There is much better hardware to use if you don't care about blobs. 

> I just refuse to sacrifice a usb port

https://github.com/SolidHal/AsusC201-usb-wifi-from-webcam solves this :)

closed 2022-10-21 by SolidHal