SolidEva/PrawnOS #305: Asus C201p Sound Issue

Wooty-B, 2022-09-15:
I have been playing around with PrawnOS and Cadmium, trying to understand the devices on this platform. When I installed the most recent version of PrawnOS, I noticed sound was not working out-of-the-box and started browsing these two issue trackers. Need to see if other C201p owners are having sound issues on PrawnOS, and if so, the below is the remedy to this (Thanks, @peter-marshall5!!!):

```
cd /usr/share/alsa/ucm2
cp Rockchip/max98090/ VEYRONI2S/ -r
mv VEYRONI2CS/max98090.conf VEYRONI2S/VEYRONI2S.conf
```
If the above doesn't start working after a reboot:

`nano /usr/share/alsa/usm2/VEYRONI2S/VEYRONI2S.conf`

```
Syntax 2

Comment "Rockchip Veyron card"

SectionUseCase."HiFi" {
        File "/VEYRONI2S/HiFi.conf"
        Comment "Default"
}

```

cocojamb0, 2022-10-20:
I've been experiencing a certain form of this issue myself - audio through headphones but not through the speakers.
For info, I'm using i3 window manager and rather than one of the graphical environments offered during the PrawnOS installation, and pulse for audio.

I'll try your fix when I'm home - thank you for adding this information.

Wooty-B, 2022-10-20:
Don't hesitate to update with your results. I will be actively working on several ARM projects this weekend and more than happy to help troubleshoot through this.

cocojamb0, 2022-10-21:
Hi again - I tried out your fix! I'm still not getting audio through  the speakers unfortunately, although the audio through the 3.5mm jack connector is still a-ok and capable of driving my Beyerdynamic DT-150 headphones

gregordinary, 2022-10-23:
Just downloaded and installed the latest pre-built PrawnOS image. 

1. I can confirm I have no sound through the speakers (Installed default XFCE). 
2. I know speakers worked previously so this shouldn't be a hardware issue. 
3. I tried both fixes from @Wooty-B - Unfortunately still no sound. 

I also looked through the [InstallingDebianOn](https://wiki.debian.org/InstallingDebianOn/Asus/C201#Audio) documentation and they note the upcoming Debian Bookworm ships with a UCM profile for the soundcard in the C201 (plus an [updated .conf file](https://github.com/alsa-project/alsa-ucm-conf/blob/a8796fc159d94131f032f9611005d304361edd11/ucm2/Rockchip/max98090/HiFi.conf) is available for HDMI audio), so rebasing on Bookworm or backporting some configurations might help to resolve this issue.

Stormwoodpecker, 2022-11-04:
Using the updated ucm2 conf seems to resolve the speaker issue here, may be worth manually adding it to the image until the eventual rebase to bookworm

cocojamb0, 2022-11-07:
> Using the updated ucm2 conf from the ALSA repo seems to resolve the speaker issue here, may be worth manually adding it to the image until the eventual rebase to bookworm

For my own clarity, are you copying the [conf listed above ](https://github.com/alsa-project/alsa-ucm-conf/blob/a8796fc159d94131f032f9611005d304361edd11/ucm2/Rockchip/max98090/HiFi.conf) and placing it in /usr/share/alsa/ucm2 ? 
I tried that and sadly it did not work for me, though headphone audio still worked.

Stormwoodpecker, 2022-11-07:
> For my own clarity, are you copying the [conf listed above ](https://github.com/alsa-project/alsa-ucm-conf/blob/a8796fc159d94131f032f9611005d304361edd11/ucm2/Rockchip/max98090/HiFi.conf) and placing [...]

Yes, using that config. I think the only important deviation is that I'm using mainline kernel 5.15.76 instead of the default 5.15.6.

There may have been relevant changes to ucm2 so it may be worth trying against a newer kernel ver and see if it can be repro'd

peter-marshall5, 2022-11-07:
I can confirm that the updated conf is working on Linux 6.0. Although I'm still having the issue where a reboot is required for the internal microphone to work after switching to the headset microphone.

cocojamb0, 2022-11-18:
I'll be honest, I haven't manually upgraded to a newer kernel on a system without a GRUB configuration before - how would I do that in this case? I've got a newer kernel version in the /boot sector but it's just the generic armhf kernel image rather than a PrawnOS 6.0 kernel.

Does anyone have any resources on how to achieve updating to a new kernel on PrawnOS, or the patience to instruct me in it?

peter-marshall5, 2022-11-19:
Arm Chromebooks use a bootloader called Depthcharge that's built into the system firmware. The depthcharge payload requires a specially formatted and signed kernel to be written to a certain partition I wrote a nice guide [here](https://github.com/SolidHal/PrawnOS/issues/282#issuecomment-836746297) on how to compile and boot your own blobby kernel for PrawnOS.

[This page](https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices/custom-firmware/) gives some nice information on the Depthcharge payload as well as on how to package and sign custom kernels.

You can also check out [this repo](https://github.com/peter-marshall5/linux-veyron-patches-and-apkbuild) for the files I use to build custom kernels on my Asus C201.

SvenRuin, 2023-04-26:
Thank you! The method above (using cd, cp an mv commands) worked well for me with the latest pre-built PrawnOS image, but I changed "VEYRONI2CS" to "VEYRONI2S". Watching videos on YouTube in Firefox, I can hear sound as I should.

However, when I try Jitsi Meet in Firefox, sound and built-in camera don't seem to work. Does anyone know a way to make this work? For the camera, Jitsi Meet says: "Failed to access your camera"

SvenRuin, 2023-04-26:
Perhaps the problem I wrote about before is in Jitsi Meet. Now the camera sometimes works there.

Anthony-Sensors, 2024-03-04:
After upgrading to bookworm I had no sound. Installing pipewire fixed the issue.